OPCFoundation/UA-.NETStandard #357: Support Read/Write of ExtensionObjects according to OPC UA Specification 1.04

coconut147, 2018-02-14:
OPC Specification defines a new Attribute Datatypedefinition. Currently this is not supported but is needed for new Servers.

mregen, 2018-02-15:
Hi @coconut147, you basically ask to switch core to the Nodeset 1.04... please stay tuned, this is work in progress.

mregen, 2018-02-20:
-->  #363

coconut147, 2018-02-28:
Hey @mregen - Any updates or timeschedule regarding this topic!? I'm also interested in reading the new Attributes (e.g. AccessLevelEx)

AlinMoldovean, 2018-02-28:
Hi @coconut147 ,

Support for reading AccesLevelEx should be included  in commit 0894364ca94d20f298e8480cb871fd76c2190ba6

coconut147, 2018-02-28:
Awesome - any plans for reading / writing structures!?

AlinMoldovean, 2018-03-07:
Support for decoding custom structures is on the long list of feature requests, but for the moment there is no time schedule for it.

coconut147, 2018-03-21:
--> #410 

Closing this issue as it is treated as Feature Request.

closed 2018-03-21 by coconut147