OPCFoundation/UA-.NETStandard #361: System.ArgumentException: invalid PFX version

Shibynator, 2018-02-16:
Hi folks. 

When I start the Xamarin Android Sample I get following Exception:

`System.ArgumentException: invalid PFX version`

from following call:

`X509Certificate2 certificate = new X509Certificate2(file.FullName);`

FullName = /storage/emulated/0/OPC Foundation/PKI/own/certs/UA Xamarin Sample Client [C938BBBE35F21F981FA07999658A7E90C13AEE87].der

Stacktrace:

```
 	[External Code]		Annotated Frame
>	0x80 in Opc.Ua.DirectoryCertificateStore.LoadPrivateKey at C:\Code\UA-.NETStandard-master\Stack\Opc.Ua.Core\Security\Certificates\DirectoryCertificateStore.cs:284,21	C#	Symbols loaded.
 	0x7A in Opc.Ua.CertificateIdentifier.LoadPrivateKey at C:\Code\UA-.NETStandard-master\Stack\Opc.Ua.Core\Security\Certificates\CertificateIdentifier.cs:171,21	C#	Symbols loaded.
 	0x91 in Opc.Ua.ApplicationConfiguration.Validate at C:\Code\UA-.NETStandard-master\Stack\Opc.Ua.Core\Stack\Configuration\ApplicationConfiguration.cs:418,13	C#	Symbols loaded.
 	0x132 in Opc.Ua.ApplicationConfiguration.Load at C:\Code\UA-.NETStandard-master\Stack\Opc.Ua.Core\Stack\Configuration\ApplicationConfiguration.cs:349,17	C#	Symbols loaded.
 	0x62 in Opc.Ua.Configuration.ApplicationInstance.LoadAppConfig at C:\Code\UA-.NETStandard-master\SampleApplications\SDK\Opc.Ua.Configuration\ApplicationInstance.cs:862,17	C#	Symbols loaded.
 	0x32 in Opc.Ua.Configuration.ApplicationInstance.LoadApplicationConfiguration at C:\Code\UA-.NETStandard-master\SampleApplications\SDK\Opc.Ua.Configuration\ApplicationInstance.cs:894,13	C#	Symbols loaded.
 	0xF8 in XamarinClient.SampleClient.CreateCertificate at C:\Code\UA-.NETStandard-master\SampleApplications\Samples\XamarinClient\XamarinClient\SampleClient.cs:88,17	C#	Symbols loaded.
 	0x6 in XamarinClient.MainPage.<OnConnect>b__5_0 at C:\Code\UA-.NETStandard-master\SampleApplications\Samples\XamarinClient\XamarinClient\MainPage.xaml.cs:70,42	C#	Symbols loaded.
```

dacolgit, 2018-02-19:
Hi @Shibynator ,

Could you check the Xamarin version you are using?
You should have:

Xamarin   4.8.0.760 (fc93f3f5b)
Visual Studio extension to enable development for Xamarin.iOS and Xamarin.Android.

Xamarin Designer   4.8.188 (c5813fa34)
Visual Studio extension to enable Xamarin Designer tools in Visual Studio.

Xamarin.Android SDK   8.1.5.0 (HEAD/75f8c6838)
Xamarin.Android Reference Assemblies and MSBuild support.

Xamarin.iOS and Xamarin.Mac SDK   11.6.1.4 (db807ec)
Xamarin.iOS and Xamarin.Mac Reference Assemblies and MSBuild support.

Shibynator, 2018-02-19:
I have exactly the versions you mentioned installed.

dacolgit, 2018-02-19:
It seems that your certificate file is corrupted or missing.

Check if in /OPC Foundation/PKI/own/certs/ there is the 'UA Xamarin Sample Client[xxxx] .der' file.
If there is the file delete it and execute again the sample.
If there is no file verify in 'Android Manifest' that 'WRITE_EXTERNAL_STORAGE' is checked.

Shibynator, 2018-02-20:
File was there. But Delete didn't help. I also tried with a complete new emulator instance or with wiped UserData, nothing helped.  Permission is checked.

Maybe it's a RaceCondition failure when creating/reading the pfx-file.

When I tried to run with a slower emulator with Marshmallow a SharingViolationException occured on first run. This because `CreateCertificate()` in `OnConnect()` at MainPage.xaml.cs is not correctly awaited. Well at least for my opinion.

`await Task.Run(() => OpcClient.CreateCertificate());`

Because `CreateCertificate()` is an async void Method, the new created Task (`Task.Run`) is completed even `CreateCertificate()` is not. Thats because an async void Method yields back on await. 

In my opinion the correct way to do it would be:

`public async Task CreateCertificate()`

-> Then its awaitable

`await OpcClient.CreateCertificate();`

Unfortunately that didn't solve my problem.

dacolgit, 2018-02-21:
Thanks, [Here](https://github.com/dacolgit/UA-.NETStandard) you can find the race condition fix.
Try to delete on the device the folder 'OPC Foundation'. This force the app to recreate the certificate.
Just deleting the .der file as I previously suggest you create an inconsistency between private and public key.

Shibynator, 2018-02-22:
Ok thanks to the fix. 

I already tried to delete folder. Doesn't matter.

But today I deleted my local files and cloned the repo to made a fresh start. Then I realized that the exceptions are thrown because the first thing I normaly do is to tick all Exceptions in the ExceptionSettings. So the sample isn't crashing because the exceptions are catched. But I am not sure if that supposed to be, specially when it indicates that something is wrong with the pfx file.  I guess you can verify these Excepions if you apply the same ExceptionSettings. When these Exceptions are "normaly" then I guess this can be closed.

[image]

mregen, 2018-02-23:
@Shibynator if you see the `invalid PFX version` that might be ok, we have to try different parameters to load it to make sure the private key exist and can be used. At this point an exception is possible and ok.
However there is indeed the async call not properly awaited creating the certs (rather loading the config), thats a good catch and should be fixed with @dacolgit branch.
Cheers,
Martin

Shibynator, 2018-02-23:
Ok great thx. Your are welcome.:-)

closed 2018-02-23 by Shibynator