OPCFoundation/UA-.NETStandard #364: Nodeset files from UaModeler

kapsl, 2018-02-20:
Hi,
as many others I have the problem, that we want to use the Ua-.NET Stack, but we also want to use our already existing NodeSet.xml files that we created over UaModeler to represent our models. Is it currently possible to import those Models into the .Net Stack???

cpipero, 2018-02-20:
Hello,
you will find a few examples on how to use nodeset files. You can look at the `Opc.Ua.Samples`  and the `Workshop` examples, especially the ones with the Boiler demo, which you can also see in UaModeler.

The second question (which might grant a separate thread) has to deal with the fact that a ModelDesign file is just one way to create a NodeSet file through the use of the ModelCompiler. It is not a competing format. ModelDesign files could technically by types by hand as they are more "humanly readable". My team even created a tool (we'll be posting it on GitHub this week) that manages that through a GUI...

kapsl, 2018-02-21:
Hi and thanks for your answer!
I still don't get it completeley: In the Workshop Bioler examples in BoilerNodeManager.cs a binary .uanodes file is loaded?! Actually i nowhere see a function that loads the Nodeset.xml?!

Additionally: How can the codegeneration from Nodset.xml to the .cs classes be done? E.g. with the Prosys Stack there comes a codegenerator - but I guess the ModelCompiler can only compile the ModelDesign files?

cpipero, 2018-02-21:
I apologize, I did not look at the Boiler code closely. The .uanodes file is a binary representation of the nodeset. The `LoadPredefinedNodes()` method can read binary and XML files. It is defined in the `\Sample\Opc.Ua.Samples\CustomNodeManager.cs` file as a member of the `NodeStateCollection` class. You can see all the overloads in there. There are also functions to save the address space as nodeset.

On the second point you are correct, the model compiler (we have the OPC Foundation one on Github, but all the major SDK vendors have their adaptation) will create the nodeset files and generate the code: for the public model compiler, you have the option to generate C# or ANSI C code, the ProSys compiler will generate the Java classes.

AlinMoldovean, 2018-02-21:
Hello @kapsl 

The [Model Compiler tool](https://github.com/OPCFoundation/UA-ModelCompiler) generates NodeSet.xml files from "ModelDesign" files (in UAModelDesign.xsd format).
It is only a command line tool but there are also some other tools from SDK vendors with GUI.
The standard output for all tools is NodeSet xml files respecting UANodeSet.xsd schema.

There are no competing formats.
"ModelDesign" in only the input format of ModeCompiler tool, but the official format for files containing sets of OPC UA nodes is "UANodeSet.xsd".

This "UA-.NETStandard" stack has support for loading nodes from a UANodeSet files, but it requires some coding.
The NodeStateCollection.LoadFromXml() only loads from xmls endcoded as ListOfNodeState, so it does not do the job you need.

For loading from a "UANodeSet" file a complete example can be found in the SampleServer project from Softing OPC UA .NET Standard Toolkit (based on the stack from this repository).
In NodeSetImportNodeManager.ImportNodeSet(ISystemContext context, string filePath)
Something like this:

`
        _private XmlElement[] LoadFromNodeSet2Xml(ISystemContext context, string filePath, bool updateTables, List<string> namespaceUris, NodeStateCollection predefinedNodes)
        {
            if (filePath == null) throw new ArgumentNullException(nameof(filePath));_

            byte[] readAllBytes = File.ReadAllBytes(filePath);
            MemoryStream istrm = new MemoryStream(readAllBytes);

            if (istrm == null)
            {
                throw ServiceResultException.Create(StatusCodes.BadDecodingError, "Could not load nodes from resource: {0}", filePath);
            }

            return LoadFromNodeSet2(context, istrm, updateTables, namespaceUris, predefinedNodes);
        }`

`
_private XmlElement[] LoadFromNodeSet2(ISystemContext context, Stream istrm, bool updateTables, List<string> namespaceUris, NodeStateCollection predefinedNodes)
        {
            UANodeSet nodeSet = UANodeSet.Read(istrm);_

            if (nodeSet != null)
            {
                // Update namespace table
                if (updateTables)
                {
                    if (nodeSet.NamespaceUris != null && context.NamespaceUris != null)
                    {
                        for (int ii = 0; ii < nodeSet.NamespaceUris.Length; ii++)
                        {
                            context.NamespaceUris.GetIndexOrAppend(nodeSet.NamespaceUris[ii]);
                            namespaceUris.Add(nodeSet.NamespaceUris[ii]);
                        }
                    }
                }

                // Update server table
                if (updateTables)
                {
                    if (nodeSet.ServerUris != null && context.ServerUris != null)
                    {
                        for (int ii = 0; ii < nodeSet.ServerUris.Length; ii++)
                        {
                            context.ServerUris.GetIndexOrAppend(nodeSet.ServerUris[ii]);
                        }
                    }
                }

                // Load nodes
                nodeSet.Import(context, predefinedNodes);

                return nodeSet.Extensions;
            }

            return null;
        
`

closed 2018-03-07 by AlinMoldovean

Houve, 2018-04-09:
What's the purpose of the **`** ? The code is really unreadable with quoting and simple text in a mix... :(

AlinMoldovean, 2018-04-10:
Hi @Houve ,

The code part was put like that by copy/paste.
I made an update and it should be better now.