OPCFoundation/UA-.NETStandard #366: Opc.Ua.CertificateGenerator Executable Dependency in Latest NuGet Libraries

RahulKate, 2018-02-21:
We need some information about Opc.Ua.CertificateGenerator executable or utility,

We are using latest Opc.Ua libraries, which was downloaded from NuGet package manager.

We found some dependency with Opc.Ua.CertificateGenerator executable in earlier version (not from NuGet) of OpcUa libraries, same utility require to generate certificate @ opc ua client side.    

We observed that, OPC Foundation has removed the dependency of Opc.Ua.CertificateGenerator in latest version which is distributed from NuGet package.

Could you please explain details of the same and confirm about dependency of Opc.Ua.CertificateGenerator?

Regards,
Rahul

AlinMoldovean, 2018-02-21:
Hi @RahulKate ,

This UA-.NETStandard stack uses the BouncyCastle.Crypto.dll library (as NuGet reference) for generating certificates and it has no dependencies for external tools.

The [CertificateGenerator  ](https://github.com/OPCFoundation/Misc-Tools/tree/master/CertificateGenerator) dependency for creating application certificates is only for the [UA-.NET-Legacy](https://github.com/OPCFoundation/UA-.NET-Legacy) stack.

RahulKate, 2018-02-22:
Hi @AlinMoldovean ,

Thanks for clarification

closed 2018-02-22 by RahulKate