OPCFoundation/UA-.NETStandard #367: OPC UA Client and Server Communication Specifications

RahulKate, 2018-02-22:
We are testing communication and data exchange between OPC UA Client and Server,

We need some information regarding guidelines and specifications regarding communication of OPC UA Client and Server.

Assume both OPC UA Server and Client are running,

If configuration / namespace / data type changes for point/tag in OPC UA Server, **Do we need to restart OPC UA client** to accommodate the new namespace configuration or data type changes?. Changes in tags which OPC UA Client has subscribed towards OPC UA Server.

please redirect me towards if any specification in this regards.

Regards,
Rahul

AlinMoldovean, 2018-02-22:
Hi @RahulKate ,

In OPC UA the server address space is dynamic by design and can change during the runtime or the application.
The clients need implement all required synchronization without restarting the application or reconnecting the session.

RahulKate, 2018-03-08:
Hi @AlinMoldovean, 

We are using underlying OPC UA Foundation library (NuGet version: 1.3.350.6),

Assume both OPC UA Server and Client are running,

Could you please guide us, how OPC UA Client library application know that OPC UA Server is reconfigured with namespace / data type and client application has to synchronize with server at run-time.

How to identify changes in tags which OPC UA Client has subscribed towards OPC UA Server.

Regards,
Rahul

fchaxel, 2018-03-10:
Hi @RahulKate, 

I'm not sure it's the right way but modification in the server adress space is followed (optional? mandatory?) by a GeneralModelChangeEventState event or a BaseModelChangeEventStates event sent at the Server object level.

The GeneralModelChangeEventState event contains a Changes property, ModelChangeStructureDataType array, with all the Affected nodeId and the verb/reason (NodeAdded, NodeDeleted, ReferenceAdded, ReferenceAdded, DataTypeChanged).
The BaseModelChangeEventStates do not contains information to identify the changes.

Client could get back this event and modify its subscriptions.

Bye.

RahulKate, 2018-03-13:
Hi,

We are using OPC UA Library from Nuget for OPC UA Client application, I am not able to find any event method defined either for Session or Subscription class.

[image]

Could you please guide us, for which class we need to subscribe to get mentioned notification events in above your reply?

OR 

From where to subscribe events **GeneralModelChangeEventState / BaseModelChangeEventStates**?

Regards,
Rahul

fchaxel, 2018-03-13:
Hi,

Assume subscription is set published active

            MonitoredItem monitoredItem = new MonitoredItem();
            monitoredItem.StartNodeId = ObjectIds.Server;
            monitoredItem.AttributeId = Attributes.EventNotifier; // default is Attributes.Value, don't want it
            EventFilter filtre = new EventFilter();
            //  select clause defines all returned values if present.
            filtre.AddSelectClause(ObjectTypeIds.BaseEventType, new QualifiedName("EventType"));
            filtre.AddSelectClause(ObjectTypeIds.BaseEventType, new QualifiedName("Message"));
            filtre.AddSelectClause(ObjectTypeIds.GeneralModelChangeEventType, new QualifiedName("Changes"));

            monitoredItem.Filter = filtre;

            monitoredItem.Notification += OnAlarmEventNotification;

            subscription.AddItem(monitoredItem);session.AddSubscription(subscription);
            subscription.Create();

 

Then in the event handler

        private static void OnAlarmEventNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
        {
            EventFieldList notification = e.NotificationValue as EventFieldList;

            if (notification == null) return; // Not an Event/Alarm Notification

            Console.WriteLine("Event notification {0} : {1}", notification.EventFields[0].Value, notification.EventFields[1].Value);
        }

Check also at the index 2 when the event is a GeneralModelChange (cast to NodeId the value at index 0 for the test) : it will be an array of ModelChangeStructureDataType.

Good luck.

RahulKate, 2018-03-22:
Thanks for your reply,

We will check and inform you, in case of any issue/problem.

closed 2018-03-22 by RahulKate