OPCFoundation/UA-.NETStandard #369: Overload for ExpandedNodeId.ToNodeId method without NamespaceTable parameter

msalafia, 2018-02-22:
Currently the method ToNodeId of ExpandedNodeId requires a parameter named **namespaceTable**.
Why is not possible convert ExpandedNodeId to NodeId without providing NamespaceTable? 

**ExpandedNodeId** provides all the necessary information in order to accomplish to this task, i think that a method like 
```Csharp
public static NodeId ToNodeId(ExpandedNodeId nodeId)
```
should be provided from ExpandedNodeId class instead of require custom code from developers.

AlinMoldovean, 2018-02-23:
Hello @msalafia ,

ExpandedNodeId can be converted to NodeId without the namespace table only if the instance of the ExpandedNodeId is not absolute, i.e. NamespaceUri is null and ServerIndex is 0.
For absolute ExpandedNodeId namespaceTable is needed to resolve the NamespaceIndex of the NodeId.

closed 2018-02-23 by AlinMoldovean

msalafia, 2018-02-23:
hi @AlinMoldovean ,

but ExpandedNodeId seems containing always NamespaceIndex in a property. Im doing the conversion with an extension method of ExpandedNodeId in this way:

```Csharp
public static class ExpandedNdeIdExtension
    {
        public static NodeId ToNodeId(this ExpandedNodeId expandedNodeId)
        {
            return new NodeId(expandedNodeId.Identifier, expandedNodeId.NamespaceIndex);
        }
    }
```
Is this wrong indeed? I have tried with server's nodes with namespace index different from 0 and it works. What am i missing?

AlinMoldovean, 2018-02-23:
> return new NodeId(expandedNodeId.Identifier, expandedNodeId.NamespaceIndex);
is wrong if "expandedNodeId.IsAbsolute = true"

If the NamespaceUri property of ExpandedNodeId is set, the NamespaceIndex should be calculated from the namespace table!

msalafia, 2018-02-23:
Ok thank you for the clarification!