OPCFoundation/UA-.NETStandard #370: Reading a Node Value

MrPoudel, 2018-02-22:
Hi everybody,

I am trying to write a very simple OPC UA console client. I found the following example in the internet. This shows how to subscribe a node and works fine. What would be the code if I just want to read the node Server_ServerStatus_CurrentTime?
`using System;
using System.Collections.Generic;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Threading;

using Opc.Ua;
using Opc.Ua.Client;

namespace opcFoundationClient
{
    class Program
    {
        static void Main(string[] args)
        {

            //Task t = ConsoleSampleClient("opc.tcp://10.105.4.96:16664");

            Console.WriteLine("Step 1 - Create a config.");
            var config = new ApplicationConfiguration()
            {
                ApplicationName = "MyHomework",
                ApplicationType = ApplicationType.Client,
                SecurityConfiguration = new SecurityConfiguration { ApplicationCertificate = new CertificateIdentifier { StoreType = @"Windows", StorePath = @"CurrentUser\My", SubjectName = Utils.Format(@"CN={0}, DC={1}", "MyHomework", System.Net.Dns.GetHostName()) }, TrustedPeerCertificates = new CertificateTrustList { StoreType = @"Windows", StorePath = @"CurrentUser\TrustedPeople", }, NonceLength = 32, AutoAcceptUntrustedCertificates = true },
                TransportConfigurations = new TransportConfigurationCollection(),
                TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
                ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = 60000 }
            };
            config.Validate(ApplicationType.Client);
            if (config.SecurityConfiguration.AutoAcceptUntrustedCertificates)
            {
                config.CertificateValidator.CertificateValidation += (s, e) => { e.Accept = (e.Error.StatusCode == StatusCodes.BadCertificateUntrusted); };
            }

            Console.WriteLine("Step 2 - Create a session with your server.");
            using (var session = Session.Create(config, new ConfiguredEndpoint(null, new EndpointDescription("opc.tcp://10.105.4.96:16664")), true, "", 60000, null, null))
            {
                Console.WriteLine("Step 3 - Browse the server namespace.");
                ReferenceDescriptionCollection refs;
                Byte[] cp;
                session.Browse(null, null, ObjectIds.ObjectsFolder, 0u, BrowseDirection.Forward, ReferenceTypeIds.HierarchicalReferences, true, (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, out cp, out refs);
                Console.WriteLine("DisplayName: BrowseName, NodeClass");
                foreach (var rd in refs)
                {
                    Console.WriteLine("{0}: {1}, {2}", rd.DisplayName, rd.BrowseName, rd.NodeClass);
                    ReferenceDescriptionCollection nextRefs;
                    byte[] nextCp;
                    session.Browse(null, null, ExpandedNodeId.ToNodeId(rd.NodeId, session.NamespaceUris), 0u, BrowseDirection.Forward, ReferenceTypeIds.HierarchicalReferences, true, (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method, out nextCp, out nextRefs);
                    foreach (var nextRd in nextRefs)
                    {
                        Console.WriteLine("+ {0}: {1}, {2}", nextRd.DisplayName, nextRd.BrowseName, nextRd.NodeClass);
                    }
                }

                Console.WriteLine("Step 4 - Create a subscription. Set a faster publishing interval if you wish.");
                var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };

                Console.WriteLine("Step 5 - Add a list of items you wish to monitor to the subscription.");
                var list = new List<MonitoredItem> { new MonitoredItem(subscription.DefaultItem) { DisplayName = "ServerStatusCurrentTime", StartNodeId = "i=2258" } };
                list.ForEach(i => i.Notification += OnNotification);
                subscription.AddItems(list);

                Console.WriteLine("Step 6 - Add the subscription to the session.");
                session.AddSubscription(subscription);
                subscription.Create();

                Console.WriteLine("Press any key to remove subscription...");
               
                Console.ReadKey(true);
            }

            Console.WriteLine("Press any key to exit...");
            Console.ReadKey(true);

        }

        private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
        {
            foreach (var value in item.DequeueValues())
            {
                Console.WriteLine("{0}: {1}, {2}, {3}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
            }
        }

    }
}
`

AlinMoldovean, 2018-02-23:
Hi @MrPoudel ,

Your example is from [NetCoreConsoleClient](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/Samples/NetCoreConsoleClient/Program.cs)

You can read the Value of a variable like this:

` Console.WriteLine("5 - Read the value of the ServerStatus.CurrentTime node.");
            var readResult = session.ReadValue(VariableIds.Server_ServerStatus_CurrentTime);
            Console.WriteLine("Read result = " + readResult.Value.ToString());`

MrPoudel, 2018-02-23:
Hi @AlinMoldovean,

Thank you very much. It works fine but only for the variables inside Server Object.
How can I read the following variable outside Server Object?

datatype :           Double 
Identifier:            myLevel
NodeId:             
                            NamespaceIndex =2             IdentifierType = String       Identifier = myLevel     
Whose paths in server's address space is : Objects->MyObjects->MyDevice->MyLevel

MrPoudel, 2018-02-24:
I just found a way to read a variable node with Identifier and NamespaceIndex.
`
NodeId nodeId = new NodeId("MyLevel", 2);
DataValue readResult2 = session.ReadValue(nodeId);
Console.WriteLine("Read result = " + readResult2.Value.ToString());
`

closed 2018-02-24 by MrPoudel

AlinMoldovean, 2018-09-11:
Hi @NichtEuler,

Maybe you can use the predefined NodeIds from DataTypeIds:

```
 public static partial class DataTypeIds
    {
        /// <summary>
        /// The identifier for the BaseDataType DataType.
        /// </summary>
        public static readonly NodeId BaseDataType = new NodeId(Opc.Ua.DataTypes.BaseDataType);

        /// <summary>
        /// The identifier for the Number DataType.
        /// </summary>
        public static readonly NodeId Number = new NodeId(Opc.Ua.DataTypes.Number);

        /// <summary>
        /// The identifier for the Integer DataType.
        /// </summary>
        public static readonly NodeId Integer = new NodeId(Opc.Ua.DataTypes.Integer);

        /// <summary>
        /// The identifier for the UInteger DataType.
        /// </summary>
        public static readonly NodeId UInteger = new NodeId(Opc.Ua.DataTypes.UInteger);

        /// <summary>
        /// The identifier for the Enumeration DataType.
        /// </summary>
        public static readonly NodeId Enumeration = new NodeId(Opc.Ua.DataTypes.Enumeration);

        /// <summary>
        /// The identifier for the Boolean DataType.
        /// </summary>
        public static readonly NodeId Boolean = new NodeId(Opc.Ua.DataTypes.Boolean);

        /// <summary>
        /// The identifier for the SByte DataType.
        /// </summary>
        public static readonly NodeId SByte = new NodeId(Opc.Ua.DataTypes.SByte);
.....
```

NichtEuler, 2018-09-12:
Hi,

Thank you so very much @AlinMoldovean 
I came up with this solution:
```
VariableNode variableNode = m_session.NodeCache.Find(expandedNodeId) as VariableNode;
uint i = (uint)variableNode.DataType.Identifier;
string browseName= DataTypes.GetBrowseName((int)i);
```
Works as it's supposed to, but correct me if I'm wrong.