OPCFoundation/UA-.NETStandard #372: OP UA Certificate

RahulKate, 2018-02-23:
Hi OPC Foundation team, 

How to generate certificate that are required to execute OPC UA client?

Is there any open tool available, Can we generate from IIS?

Certificate generated from IIS is ready to use with OPC UA client or not.

Regards,
Rahul

AlinMoldovean, 2018-02-23:
Hi @RahulKate ,

For client and server application from this repository, the certificates are automatically created if not present in the specified configuration location: ApplicationConfiguration.SecurityConfiguration.ApplicationCertificate
For the moment BouncyCastle library is used for that.

You can also use certificates issues by Certificate Authorities or create them with [CertificateGenerator](https://github.com/OPCFoundation/Misc-Tools/tree/master/CertificateGenerator)

RahulKate, 2018-03-10:
Thank you for reply

closed 2018-03-10 by RahulKate