OPCFoundation/UA-.NETStandard #373: OPC UA Live Data Events [Question]

eltigre4jc, 2018-02-23:
Hello!

I have built an application for VB .NET to test a OPC server that we have in our devices. 
And I have implement only a handler for Session.Keep_Alive function and for subscription notification.
We want too see as many details as possible about the connection, server status, disconnect reasons, error messages etc.

My question is: Are they any more events that I need to subscribe to? Maybe Session.Notification? 
Here [https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/](url) It says for Jave, about ServerStatusListener . Is there anything in .NET like that?

Also, can I set somewhere at which interval the keep alive function will be called?

Thank you

AlinMoldovean, 2018-02-23:
Hi @eltigre4jc ,

You need to use the session.KeepAlive event.
The pooling interval of keealive read can be changed from session.KeepAliveInterval
Default value is 5000 (5 seconds)

eltigre4jc, 2018-02-23:
Ok.. Thank you. Didn't expect to only handle one event :)

closed 2018-03-09 by mregen