OPCFoundation/UA-.NETStandard #374: Problem from Windows to Mono

fchaxel, 2018-02-25:
Hi,

I have a Server code just ported from UA-.NET-Legacy to UA-.NETStandard.

It's now a .NET Fw 4.6 application working like a charm on Windows.

I'm trying to run it on mono (Rpi 3) : copy .exe & core.dll,  server.dll, configuration.dll & BouncyCastle.dll . But some problems occured.

The PKI/own directory is automatically created by the code (but without certs & private sub directories), and I have the following crach during the call to CheckApplicationInstanceCertificate

Exception: Creating X509Certificate from PKCS #12 store failed ---> System.Security.Cryptography.CryptographicException: The public/private key pair in the certficates do not match.

The keys are created by the UA stack : the time is from long to quite long before the error occur. But nothing in the sub-directories.

If I puts by hand the two files .der & pfx in the sub directories (coming from Xca on Windows), the code runs speed (not key creation sure) but I also got a crach :

Exception: Specified cast is not valid.
  at Opc.Ua.CertificateValidator.Validate (System.Security.Cryptography.X509Certificates.X509Certificate2Collection chain)

Somebody with an idea ?

Thanks.

fchaxel, 2018-02-26:
Hi,

A beginner's problem ! Mono version was very old 4.9.2. Now with the 5.8 all is OK (45 minutes to install it).

Bye

mregen, 2018-02-27:
@fchaxel sidenote: X509Store is not supported on mono (doesn't save pfx files).
For command line apps there is also .Net Core support on Pi...
Cheers,
Martin

closed 2018-03-07 by fchaxel