OPCFoundation/UA-.NETStandard #379: QueryFirst example

thutch, 2018-03-01:
Is there an example or some sample available that demonstrates how to use the QueryFirst service?  
Wanting to query for all objects that HasTypeDefinition of AlarmConditionType.
Similar queries for Variables HasTypeDefinition of BaseDataVariableType
Also for Events, but excluding Alarms.

AlinMoldovean, 2018-03-07:
Hi @thutch ,

An option can be to make a backward Browse on the with "HasTypeDefinition" reference on the type node.
But that will return only the nodes that have bidirectional HasTypeDefinition references.

thutch, 2018-03-08:
Thank you for the response. 
For the variables, am looking at Browsing, then for each reference Browsing for the References and checking them for their HasTypeDefinition.

For the events, it looks like would need to test if there is an EventNotifier attribute value set to 1 for a Node that is an Object.  

The goal is to be able to pro-grammatically collect all the alarms, events (or the objects they notify) and variables that can be monitored.  

What would be the approach for doing this?

zlatinejc, 2018-04-19:
What is the purpose of reference browsing, why not browse nodes per se recursively ?

closed 2018-06-14 by AlinMoldovean