OPCFoundation/UA-.NETStandard #381: Connect to OPC UA Server with UserName/Password

sanny32, 2018-03-02:
What is the way to configure OPC UA Server to connect with UserName and Password?
I'm tried this configuration in server xml file configuration, but without success.
`
 <RegistrationEndpoint>
     <ua:UserIdentityTokens>
        <ua:UserIdentityToken>
          <ua:UserName>User1</ua:UserName>
          <ua:Password>1</ua:Password>
        </ua:UserIdentityToken>
     </ua:UserIdentityTokens>
</RegistrationEndpoint>
`

fchaxel, 2018-03-04:
Hi @sanny32 

For a full code see in UA-.NETStandard\SampleApplications\Workshop\UserAuthentication

In the xml file section UserTokenPolicies
`<ua:UserTokenPolicy>
        <ua:TokenType>UserName_1</ua:TokenType>
      </ua:UserTokenPolicy>`

means that a client could provided a username with (or without) password for login. It's not the place to put a login/password list.

With `<UserTokenPolicy>` Anonymous and X509 personnal certificate : remove them if you just want login/password.

For login/password in the child class of `StandardServer` adds something like the following code. 

`protected override void OnServerStarted(IServerInternal server) 
{
            base.OnServerStarted(server);
            server.SessionManager.ImpersonateUser += new ImpersonateEventHandler(SessionManager);
        }`

SessionManager just creates the UserIdentity object to accept the user ... or not to reject it.
Inside it login/pass are verified using userNameToken.UserName, userNameToken.DecryptedPassword properties.

` private void SessionManager(Session session, ImpersonateEventArgs args)
        {
            UserNameIdentityToken userNameToken = args.NewIdentity as UserNameIdentityToken;
            if (userNameToken != null)
            {                
                args.Identity = new UserIdentity(userNameToken);
                return;
            }
        }`

Bye.

closed 2018-03-07 by AlinMoldovean

sanny32, 2018-03-07:
Exploring the source code, I have found the user account path in OS: %CommonApplicationData%\OPC Foundation\Accounts\{Application Name}\{UserName}.xml, where {Application Name} - name of the executable file of the OPC UA Server, {UserName} - name of the user account.
This xml file contains user name and encrypted password.
To create this file you can use Opc.Ua.ConfigurationTool.exe or UserNameCreator class in the Opc.Ua.Configuration namespace (UA COM Interop Library).
This way allows to use Username/Password acces to OPC UA COM Server Wrapper.