OPCFoundation/UA-.NETStandard #382: Error in latest OPC UA source code, for communication with https endpoints

RahulKate, 2018-03-06:
Hi,

We are using latest OPC UA Source code with version **"UA-.NETStandard-master 1.03.351.7"**.

We found below issue while connecting OPC Foundation's Data Access Server using Data Access Client with below endpoints,

**https://desktop-machine:62546/Quickstarts/DataAccessServer/** 

Issue as,
[image]

Expecting OPC Foundation team about resolution of the same issue and inform us if OPC UA stack is supporting on https endpoints or not.

Regards,
Rahul

mregen, 2018-03-07:
@RahulKate good catch, the quickstart servers all lack the secret sauce in the project files to get the libuv.dll copied to the release folders.. --> #359

RahulKate, 2018-03-08:
Hi @mregen ,

Thanks for pointing to solutions, it would be difficult to understand treads (#359) and came out of problems.

We are using below library/project types,
1.  library **"UA-.NETStandard-master 1.03.350.6" - From NuGet**
2.  Visual Studio 2015

We developed OPC UA client based on OPC UA library, Can we able to communicate between OPC UA Server and Client over **https** endpoints using the above mentioned library and projects?

What are the changes we need to to in our project?

We need to check its working over https endpoint or not, what could be the correct and easiest way to check. 

Could you please guide us with https communication over OPC UA Server and Client? do we need to get actual CA certificate?
Can we use self signed certificate? What could be the procedure? 

Please guide in right direction.....

Regards,
Rahul

mregen, 2018-03-08:
Hi @RahulKate, the https project files samples are fixed. --> #386. 
1.03.350.6 has known issues with https, please wait for the next nuget packages and use the v1.03 or master branch with VS 2017 for now.

Currently the https server uses the self signed application cert of the UA server and the validator of the UA client. https cert store is on the roadmap.

Cheers,
Martin

RahulKate, 2018-03-09:
Hi @mregen ,

Thanks for explanation, 

Latest NuGet library avaiable is **OPCFoundation.NetStandard.Opc.Ua 1.3.351.7**,

Can we use NuGet Version (1.3.351.7) for **https** communication, or still we need to wait for next release?

Regards,
Rahul

mregen, 2018-03-09:
Sorry @RahulKate, you have to wait for the next release or build your private nuget in the meantime (template in /nuget) and consume private packages in your project. (See comment in Nuget.Config how to consume nuget packages from a local feed.)

Cheers,
Martin

RahulKate, 2018-03-09:
Hi @mregen ,

Thanks for your reply,

We will wait for another new NuGet packages.

Regards,
Rahul

closed 2018-03-10 by RahulKate