OPCFoundation/UA-.NETStandard #384: How to set up server to only accept trusted users?

NichtEuler, 2018-03-07:
Hey,

I want my server my server to only accept registered users.
How am can I:

1. Set up a new user with a password.
2. Get the server to only connect when this specific user is loging in?

This would help me a great deal.
Thanks in advance.

fchaxel, 2018-03-07:
Hi @NichtEuler 

In the xml file section UserTokenPolicies
<ua:UserTokenPolicy> <ua:TokenType>UserName_1</ua:TokenType> </ua:UserTokenPolicy>

means that a client could provided a username with (or without) password for login.

<UserTokenPolicy> Anonymous and X509 personnal certificate : remove them if you just want login/password.

Bye

AlinMoldovean, 2018-03-07:
For _"2. Get the server to only connect when this specific user is loging in?"_ you can use SampleServer as an example.
Set the config like this:
```
    <UserTokenPolicies>
      <!-- Allows username/password -->
      <ua:UserTokenPolicy>
        <ua:TokenType>UserName_1</ua:TokenType>
        <!-- passwords must be encrypted - this specifies what algorithm to use -->
        <ua:SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256</ua:SecurityPolicyUri>
      </ua:UserTokenPolicy>

    </UserTokenPolicies>

```

And in SampleServer.UserAuthentication.cs.SampleServer.VerifyPassword(string userName, string password) put a custom logic to decide what userName/pasword combination is allowed.
e.g.

```
        /// <summary>
        /// Validates the password for a username token.
        /// </summary>
        private void VerifyPassword(string userName, string password)
        {
            if (userName != "user1" && password != "password1")
            {
                // construct translation object with default text.
                TranslationInfo info = new TranslationInfo(
                    "InvalidPassword",
                    "en-US",
                    "Specified password is not valid for user '{0}'.",
                    userName);

                // create an exception with a vendor defined sub-code.
                throw new ServiceResultException(new ServiceResult(
                    StatusCodes.BadIdentityTokenRejected,
                    "InvalidPassword",
                    "http://opcfoundation.org/UA/Sample/",
                    new LocalizedText(info)));
            }
        }
```

NichtEuler, 2018-03-07:
Thank you both for the help, this is much appreciated.
Got this working.

closed 2018-03-07 by NichtEuler