OPCFoundation/UA-.NETStandard #388: SetTriggering, Event triggeringItem, variables links.  Not notified of all linked variables

thutch, 2018-03-08:
Am wanting to use the SetTriggering to collect variable values when an event happens. 
Have put together a sample that works against the Quickstarts.SimpleEventsServer.   It creates a MoniteredItem on an event, and two more MonitoredItems against variables. 
One of the variables does not change.  Another variable whose value changes based upon time (from another SDK sample). 
Found that when the Event fires, the MonitorredItem with the variable that is changing regularly notifies of the change.  There is no notification on the MonitoredItem with the variable that has not changed.
My expectation was to be notified of all the variable values, regardless of whether the variable value has changed or not, using the SetTriggering.

Is that the expected behavior, to not be notified by a MonitoredItem with a variable when the value of the variable has not changed, as part of the SetTriggering?
 
Or have I done something incorrectly?

The SetTriggering statusCodes are showing Good for both linked MonitoredItems with the variables.

Here's the code that sets up the SetTriggering;
```
var subscription = new Subscription(session.DefaultSubscription) { PublishingInterval = 1000 };

MonitoredItem monitoredItem = new MonitoredItem(subscription.DefaultItem);

monitoredItem.StartNodeId = 2259;
monitoredItem.AttributeId = Attributes.Value;
monitoredItem.DisplayName = "State";
monitoredItem.MonitoringMode = MonitoringMode.Sampling;
monitoredItem.SamplingInterval = 1000;
monitoredItem.QueueSize = 0;
monitoredItem.DiscardOldest = true;

var monitoredItem2 = new MonitoredItem(subscription.DefaultItem)
{
	DisplayName = "ServerStatusCurrentTime",
	StartNodeId = "i=" + Variables.Server_ServerStatus_CurrentTime.ToString(),
	MonitoringMode = MonitoringMode.Sampling,
	SamplingInterval = 1000,
	QueueSize = 0,
	DiscardOldest = true
};

Console.WriteLine("5b - Define the event fields you want to receive.");
var selectClause = new[] {
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventId") }, AttributeId = Attributes.Value },
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("EventType") }, AttributeId = Attributes.Value },
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("SourceName") }, AttributeId = Attributes.Value },
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Time") }, AttributeId = Attributes.Value },
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Message") }, AttributeId = Attributes.Value },
	new SimpleAttributeOperand { TypeDefinitionId = ObjectTypeIds.BaseEventType, BrowsePath = new[] { new QualifiedName("Severity") }, AttributeId = Attributes.Value }
};

var filter = new EventFilter() {SelectClauses = selectClause};

var triggeringItemId = new MonitoredItem(subscription.DefaultItem)
{
	NodeClass = NodeClass.Object,
	StartNodeId = ObjectIds.Server,
	AttributeId = Attributes.EventNotifier,
	MonitoringMode = MonitoringMode.Reporting,
	SamplingInterval = -1,
	QueueSize = 100,
	CacheQueueSize = 100,
	Filter = filter
};

var list = new List<MonitoredItem>
{
	triggeringItemId,
	monitoredItem,
	monitoredItem2
};

list.ForEach(i => i.Notification += OnNotification);
subscription.AddItems(list);
session.AddSubscription(subscription);

subscription.Create();

UInt32Collection linksToAdd = new UInt32Collection(3);
linksToAdd.Add(monitoredItem.Status.Id);
linksToAdd.Add(monitoredItem2.Status.Id);

StatusCodeCollection statusCodes;
DiagnosticInfoCollection diagnosticInfos;
StatusCodeCollection statusCodesRemoved;
DiagnosticInfoCollection diagnosticInfosRemoved;
ResponseHeader responseHeader = session.SetTriggering(null, subscription.Id, triggeringItemId.Status.Id, linksToAdd,
	null,
	out statusCodes,
	out diagnosticInfos,
	out statusCodesRemoved,
	out diagnosticInfosRemoved);
```
Here's the notification handling code;
```
private static void OnNotification(MonitoredItem item, MonitoredItemNotificationEventArgs e)
{
	Console.WriteLine("MonitoredItem update: Begin");
	Console.WriteLine("DequeueEvents");
	foreach( var value in item.DequeueEvents() )
	{
		Console.WriteLine("{3}: {2}, {4}, {5}", value.EventFields.Select(v => v.Value ?? "<null>").ToArray());
	}

	Console.WriteLine("DequeueValues");
	foreach( var value in item.DequeueValues() )
	{
		Console.WriteLine("{0}: {1}, {2}, {3}", item.DisplayName, value.Value, value.SourceTimestamp, value.StatusCode);
	}
	Console.WriteLine("MonitoredItem update: End");
	Console.WriteLine("----------------------");
}
```

AlinMoldovean, 2018-03-09:
Hi @thutch ,

SetTriggering mechanism will send notificationc only for the linked monitored items that have queued data changes and not for the static values.

closed 2018-03-09 by AlinMoldovean

thutch, 2018-03-09:
Thank you for the response.  👍