OPCFoundation/UA-.NETStandard #390: OPC UA Server reconnect problems on Linux

alexandrudicu, 2018-03-09:
I have encountered some problems on Linux when using OPC UA Server and .NET Core 2.0:

- This can be reproduced with OPC UA servers created with OPCFoundation UA-NETStandard. For example, NetCoreConsoleServer project.
- Clients can be Windows or Linux OPC UA Clients and it doesn't matter if the clients use .NET Framework or .NET Core.
 
The problem is:
- On Ubuntu 16.04.4 LTS - If the server (NetCoreConsoleServer) is stopped while there are connected clients, and then started again without waiting at least one minute, any connected client will not be able to reconnect to the same restarted server and new clients will not be able to connect at all. If we wait 1 minute and then start the server, the existing clients can reconnect and new clients can connect.
- On Debian 9.3 - If the server (NetCoreConsoleServer) is stopped while there are connected clients, and then started again, any connected client will not be able to reconnect and new clients can't connect at all. On Debian, even if we wait more minutes and start the server again, existing clients won't reconnect anymore.

mregen, 2018-03-12:
Hi @alexandrudicu, I have seen this problem (on travis) before but I don't have a solution. It looks as if the port lingers in TIME_WAIT state in your case. After a timeout the port is closed. Within the port being in this wait state, the server can't reopen the listening port.
In my case I experienced the problem only when the same process tried to restart a server and the listening port. 
It may also happen that after restart the server can only reopen the Ipv4 or the Ipv6 port which would explain why you can never reconnect.
Logs?

alexandrudicu, 2018-03-12:
On the server side, the log is empty. On the client side:
Id: BadTcpInternalError
Description: An unexpected error occurred while connecting to the server.
Id: Bad
Description: No connection could be made because the target machine actively refused it.

Yes, you are correct, it is the TIME_WAIT state scenario.

AlinMoldovean, 2018-05-17:
No possible fix on Ua stack. Linux OS related issue.

closed 2018-05-17 by AlinMoldovean