OPCFoundation/UA-.NETStandard #394: NO_HTTPS on Opc.Ua.Core does not work

JensNordenbro, 2018-03-12:
I expected there to be no dependencies to assemblies    "Microsoft.AspNetCore.Server.Kestrel" and "Microsoft.AspNetCore.Server.Kestrel.Https" since NO_HTTPS conditional was activated on the project. 

This results in publishes being much bigger than they could be. 

Is this work in progress or just dead code? There is just one place where #if !NO_HTTPS is used in the code.

mregen, 2018-03-13:
Please try this branch--> https://github.com/mregen/UA-.NetStandardLibrary/tree/v103_nohttps. If it works for you we can open a PR.

JensNordenbro, 2018-03-13:
So I pulled the branch at home and it failed during build. 

The error is: "The name 'CreateHttpsServiceHost' does . not exist in the current context	Opc.Ua.Server(netstandard1.3), Opc.Ua.Server(netstandard2.0)"

Visual Studio shows the call to "CreateHttpsServiceHost"-method as not active code inside StandardServer.cs since NO_HTTPS is defined at the Project level; still the compiler complains, 

The conditional NO:HTTPS was defined on Opc.Ua.Server.csproj  as well as Opc.Ua.Core.csproj. 

Am I doing something wrong?

mregen, 2018-03-13:
@JensNordenbro a clean and rebuild should do the job. Sounds like the server assembly hasn't picked up the NO_HTTPS define. You need to uncomment the define in both csproj, as you mentioned above.

JensNordenbro, 2018-03-14:
So, I got your branch v103_nohttps_set to work. 
I have, very quickly, tried a publish of both .net core client and server and testrun and the builds produce the correct amount of dll:s in the publish folder + seem to work! 

For some strange reason I still get the same problem as mentioned above both at work and at home on the other branch. 

To conclude, I think that this is a significant improvement, although controlling HTTPS-availability by setting conditionals on many projects is not optimal. If it could be done in one location or if the opc ua endpoint setup was redisigned in a way similar to asp.net core middleware setup the code would be much easier to follow and configure. 

**Still, this is significant improvement, please do the pull Request!!**

mregen, 2018-03-15:
Hi @JensNordenbro I modified the csproj with a build flag like (in the v103_nohttps_set branch)
`dotnet build /p:NoHttps=true etcetc.` 
It allows to set the build flag on command line. Same works for msbuild. See appveyor and travis scripts for how to...  
Indeed the net core sample server publish folder size goes down from ~21MB to ~16 MB, not bad. If there are no more objections I'll open the PR.

JensNordenbro, 2018-03-15:
Great @mregen, please do! I will follow and close this issue upon PR completion.

closed 2018-03-16 by mregen

JensNordenbro, 2018-03-16:
https://github.com/OPCFoundation/UA-.NETStandard/commit/e2a40209842c067da7004503deaa20b1967e0401