OPCFoundation/UA-.NETStandard #399: Publish data change event [Question]

alvcap, 2018-03-14:
Hi!

I am writing a server that exposes some variables (type BaseDataVariableState). When I connect to the server using a client, I am able to access the variables and their values, but the client does not detect data changes on the variables when I tell him to subscribe to them. 

I suspect that I need to setup some event publishing rules, but I am struggling to find documentation about how to do it.

How can I setup data change events on my server so that a client can subscribe to data changes?

fchaxel, 2018-03-14:
Hi @alvcap 

Changes at least variable.Timespan, .StatusCode, .Value properties certainly.

forgot  variable.ClearChangeMasks(SystemContext, false) ?

Bye

alvcap, 2018-03-15:
> forgot variable.ClearChangeMasks(SystemContext, false) ?

Exactly :) Thanks!

closed 2018-03-15 by alvcap