OPCFoundation/UA-.NETStandard #402: Using LDS with sample Apps

BK-engineer, 2018-03-16:
Hi,
I am currently moving a few applications form UA.-NET-Legacy to UA.-NETStandard.

When building with UA.-NETStandard the server doesn't show up on LDS (local discovery server). This is also the case with the sample apps e.g. ReferenceServer.
I am using a local installation of LDS downloaded from opcfoundation.org version opc-ua-local-discovery-server-1.03.371.424-20171207.zip.

Any clues on this?

AlinMoldovean, 2018-03-16:
Hi @BK-engineer ,

With this "UA-.NETStandard" stack you need to set the trust relationship with LDS in order to be displayed as registered.
i.e. you need to copy the server certificate to "C:\ProgramData\OPC Foundation\UA\pki\trusted\certs"
and the LDS certificate to the trust list of the server.
You also need to set the MaxRegistrationInterval configuration parameter to a value like 30.000 (30 seconds between registrations).

The " UA.-NET-Legacy" stack contains some code in the AddToDiscoveryServerTrustList() method that was used to establish a trust relation with LDS in the background. That part was removed from this new stack.

closed 2018-03-16 by AlinMoldovean

BK-engineer, 2018-03-16:
Thank you Alin,
this helped me along!