OPCFoundation/UA-.NETStandard #409: Creating a server with minimum preconfigured tags

JensNordenbro, 2018-03-20:
I want to create a minimal server with only a few tags active. 

I have looked at the samples and found that in SampleServer one can undef CUSTOM_NODE_MANAGER in order remove the Boiler etc. 

Also there is a comment section saying:

/// Applications with small address spaces do not need to create their own NodeManagers and can add any
        /// application specific nodes to the CoreNodeManager. Applications should use custom NodeManagers when
        /// the structure of the address space is stored in another system or when the address space is too large
        /// to keep in memory.

I cant find any good examples on this. Can you please point me in a direction. 
I basically want a really lightweight example.

AlinMoldovean, 2018-05-17:
Hi @JensNordenbro , 
The recommended approach is to make your own CustomNodeManager, even if you have a small set of nodes for your application.

AlinMoldovean, 2018-05-23:
@JensNordenbro ,
Please have a look at the ReferenceServer sample:
https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/Reference