OPCFoundation/UA-.NETStandard #410: NonAtomicRead/Write Definition in AccessLevelEx is wrong

coconut147, 2018-03-21:
According to Part 3 of UA Spec 1.04 the bits 8 & 9 define the "NonatomicRead/Write" Flags.

In https://raw.githubusercontent.com/OPCFoundation/UA-.NETStandard/0894364ca94d20f298e8480cb871fd76c2190ba6/Stack/Opc.Ua.Core/Stack/Generated/Opc.Ua.DataTypes.cs

The enum AccessLevelExType is wrong and sets 

NonAtomicRead to 65536 which is bit 16 (should be 8)
NonAtomicWrite to 131072 which is bit 17 (should be 9)

mregen, 2018-03-28:
Hi @coconut147, indeed, we need to open an issue in the Nodeset repo and Mantis.

mregen, 2018-06-07:
fixed in https://github.com/OPCFoundation/UA-ModelCompiler/commit/a9699ab2f87cf4c2a01f3646010ab9189725cb5f, would be fixed here in a nodeset update

coconut147, 2018-06-08:
Any time schedule for nodeset updates - or am I able to do this as a pull request!?

AlinMoldovean, 2018-06-14:
We intend to provide this update for the nodeset files soon.

AlinMoldovean, 2018-06-26:
Fixed in Commit 00bf7b1a96b408bc2b52d95babbdf613b0cdfd7c

closed 2018-06-26 by AlinMoldovean