OPCFoundation/UA-.NETStandard #411: Communication Error: https endpoints, OPC UA Sample Client and other simulation server

RahulKate, 2018-03-22:
Hi,

We have downloaded latest OPC UA Source code on 21st March 2018, after compiling and executing OPC UA Sample Client and tries to communicate with OPC UA Sample Simulation Server (distributed by ProSys). We found below error with dialog box,

[image]

We are using OPC UA Server (distributed by ProSys), 
https://www.prosysopc.com/products/opc-ua-simulation-server/

[image]

Note: OPC UA Sample Client and OPC UA Sample Server are successfully communicating with each other over https endpoints.

OPC UA Sample Client is not working with ProSys OPC UA Sample Client. Please help us to get out of that and provide resolution for the same.

If you need further information, please reply back.

Regards,
Rahul

mregen, 2018-03-26:
@RahulKate in the screenshot you use the https prefix with the opc.tcp port 51210. But with https you should use port 51443. Is this maybe the problem?

RahulKate, 2018-03-26:
Hi,

Thanks for your response,

Currently there is issue while creating monitored items below is the error with stack trace details,

[image]

However connection was succesfull as shown in below figure,

[image]

Additionally, we also observed that, once connection established, if user is again trying to re-connecting  then its giving error as,

[image]

Please help us to resolve the same, its not reliable in latest library, what is your opinion/experience on the same?

Regards,
Rahul

AlinMoldovean, 2018-03-27:
Hi @RahulKate, 

You need to check if you have the same error when creating monitored items on the binary opc.tcp protocol instead of https in order to know if is a problem of the service request content or the transport protocol.

RahulKate, 2018-04-02:
Hi @mregen ,

Its working properly for ocp:tcp endpoints, we are looking for workable solutions on https endpoints.

Initially, we are evaluating on OPC Foundation tools/sample program, request you to guide on this issue.

Is it workable solution (https) with OPC Foundaiton library? If yes, then please help us to resolve the same issue mentioned in this thread.

Regards,
Rahul

AlinMoldovean, 2018-05-17:
Hi @RahulKate ,

Can you check also on the server vendor side for this issue?
It does not replicate with other servers on https protocol.

RahulKate, 2018-05-18:
Hi @mregen ,

Our test team is busy in other tasks, once our test team is free, we will re-open the testing activity.
For time being we can close the same thread.

Regards,
Rahul

closed 2018-05-18 by RahulKate