OPCFoundation/UA-.NETStandard #416: OPCUA Certificate Error

vrushalikatkade, 2018-03-31:
I have used the following code to read the application configuration
ApplicationInstance.MessageDlg = new ApplicationMessageDlg();

            application.ApplicationName = "Globasys";
            application.ApplicationType = ApplicationType.ClientAndServer;
            application.ConfigSectionName = "Globasys";

            // use a custom transport channel
            // WcfChannelBase.g_CustomTransportChannel = new CustomTransportChannelFactory();

            try
            {
                application.LoadApplicationConfiguration(false).Wait();

                // check the application certificate.
                bool certOK = application.CheckApplicationInstanceCertificate(false, 0).Result;
                if (!certOK)
                {
                    throw new Exception("Application instance certificate invalid!");
                }
}
catch{}

i have get the following error from the application.CheckApplicationInstanceCertificate(false, 0).Result

at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions) at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification) at System.Threading.Tasks.Task`1.get_Result() 

how to solve this error 

thanks ,
vrushali

vrushalikatkade, 2018-04-02:
@how to solve this error

AlinMoldovean, 2018-04-04:
Hi @vrushalikatkade ,

First you need to debug inside the CheckApplicationInstanceCertificate() method and check the exact condition where the exception is thrown.
It seems that your application instance certificate does not exist or is not valid.

closed 2018-04-11 by vrushalikatkade