OPCFoundation/UA-.NETStandard #418: OPC UA PerfTest Server

RahulKate, 2018-04-03:
Hi,

We are trying to test OPC UA Client application for maximum number of nodes. For maximum number of nodes (tags) we used OPC UA PerfTest Server.

OPC UA Client application uses method --> session.TranslateBrowsePathsToNodeIds() to translate string namespace address to get NodeId's for subscription.

We observed that, function call to session.TranslateBrowsePathsToNodeIds() gets blocked in case of testing with PerfTest server.

After debugging with latest source code from Github we find that below for loop from function "public virtual void TranslateBrowsePath()" is not pointing to next reference from statement "reference = browser.Next()"

So we made changes in source code as marked in yellow shades in below image, to make it work.

[image]

Requesting to OPC UA Foundation team, to check whether the same issue is in OPC UA Server library or PerftTest Server. OR Is there any configuration can resolve the same issue?

Regards,
Rahul

AlinMoldovean, 2018-04-04:
Hi @RahulKate ,

Could you explain what your fix actually does?
I only see you are assigning a "lastReference" object that is not further used..

AlinMoldovean, 2018-04-04:
Hi @RahulKate ,

Can you explain what your fix actually does?
It does not seem to be ok to stop when there are multiple references with the same target (in consecutive order).

RahulKate, 2018-04-04:
Hi @AlinMoldovean ,

Its temporary fix, to get out of infinite loop. Its not traversing to next element in case of PerfTest OPC UA Server.

We have analysed that, even after temporary fixing it hangs somewhere else.

Its not reliable to test with performance testing with PerfTest Server.

We are using ProSys OPC UA Server Simulator to test performance testing for 16000 nodes.

Regards,
Rahul

closed 2018-04-04 by RahulKate