OPCFoundation/UA-.NETStandard #422: remote OPCUA

vrushalikatkade, 2018-04-10:
I have click on browse button to discover the network PC ,this option will not display anything .
[image]
i will be search on the coding the domain name passed as null .

[image]

how to browse networking computer & discover the OPCUA server on it?

Thanks,
vrushali

mregen, 2018-04-13:
@vrushalikatkade Did you install the LDS server from the OPC Foundation web site?

vrushalikatkade, 2018-04-13:
yes, i have install the LDS 1.02

vrushalikatkade, 2018-04-16:
any suggestion ...

AlinMoldovean, 2018-04-19:
Hello @vrushalikatkade , 

This seems to be a bug in the WinForms sample control.
Can you check if remote running servers are displayed if you manually type the host name (or IP address) and click "Discover"?
The "client.FindServers()"  method of DiscoveryClient seems to work correct.

vrushalikatkade, 2018-04-20:
Hello sir@AlinMoldovean,
            Thanks for the reply,
          I have already tested the discover button by typing manual host name,but it's not working it throws exception as {"Error establishing a connection: BadNotConnected"} . "client.FindServers()" this method working for only localhost not for  remote connection.

AlinMoldovean, 2018-04-20:
Hello @vrushalikatkade ,

I also made a test and works also for remote machines and it works.
Please check on that remote machine that:
  - the LDS service is installed and running
  - there are running and registered OPC UA Servers
  - the machine and IP address is reachable
  - there is no firewall or routers blocking the communication on port 4840

vrushalikatkade, 2018-04-20:
Hello sir,
my remote computer is ubuntu 16.04(virtual machine).how to install the LDS on linux. OPCUA server is running on that machine
[image]

AlinMoldovean, 2018-04-20:
In that case you need to call GetEndpoints on the server port (instead of FindServers) or run the server on port 4840 in case it has support for FindServers (returns itself as the list of registered servers).

vrushalikatkade, 2018-04-20:
can you share any sample code for that. (but same problem for windows 7 also & in windows 7 remote OPCUA server list can not display but it will connect to the server)

vrushalikatkade, 2018-04-20:
i will change the OPCUA port to the 4840 .then try to connecting but it will display the same issue
{"Error establishing a connection: BadNotConnected"}

vrushalikatkade, 2018-04-24:
can you share your tested code (small part of the browsing OCPUA server)

AlinMoldovean, 2018-05-30:
@vrushalikatkade ,

Here is an example of using the DiscoveryClient.GetEndpoints method:

>       
        private bool DiscoverEndpoints(Uri discoveryUrl, out String message)
        {
            // use a short timeout.
            EndpointConfiguration configuration = EndpointConfiguration.Create(m_configuration);
            configuration.OperationTimeout = m_discoveryTimeout;

            DiscoveryClient client = DiscoveryClient.Create(
                discoveryUrl,
                EndpointConfiguration.Create(m_configuration));

            try
            {
                EndpointDescriptionCollection endpoints = client.GetEndpoints(null);
                OnUpdateEndpoints(endpoints);
                message = String.Empty;
                return true;
            }
            catch (Exception e)
            {
                Utils.Trace("Could not fetch endpoints from url: {0}. Reason={1}", discoveryUrl, e.Message);
                message = e.Message;
                return false;
            }
            finally
            {
                client.Close();
            }
        }

closed 2018-06-11 by AlinMoldovean