OPCFoundation/UA-.NETStandard #423: Trouble with server certificates

Matthumba, 2018-04-12:
Hello,

I am writing an OPC UA client application within an application framework that uses a quite old OPC UA SDK stack implementation (product version 1.01.333.0) on Windows 7. I tested my client against the OPCUA-.NETStandard-master reference application which I downloaded on December 14th 2017 (appveyor version 1.3.350) and also against a [Milo](https://github.com/eclipse/milo) OPC UA server  both running on the same machine as the client and it worked fine.

Recently, I updated to the newly available versions of UA-.NETStandard and Milo and now my client won't connect to either the .NetStandard or the Milo server. The problem is that on establishing a session with the server the client reports that the server's certificate is not trusted. Thereafter the certificate shows up under the untrusted certificates store saying that it has been revoked by its certificate authority. I don't have any chance to set the server's certificates as trusted, they are always being revoked. So I think something changed with the certificate handling inside the OPC UA stack of UA-.NETStandard and Milo betweeen the versions.

Unfortunately I cannot change the application framework's OPC UA stack implementation. Is there a chance that I can solve this problem anyway and how?

Any help is appreciated!

mregen, 2018-04-13:
@Matthumba have you checked your client supports most recent security profiles and key requirements? A few protocols using Sha1 have been deprecated recently and default key size is now 2048.

Matthumba, 2018-04-16:
@mregen I did some further investigation and found out that `CertCreateSelfSignCertificate` method in class `CertificateFactory` is called with a zero pointer for the `PCRYPT_ALGORITHM_IDENTIFIER` argument (be aware that I have to use an old OPCUA client library version 1.1.333.0). Thus, the client certificate is created with the default SHA1RSA signature algorithm with a 2048 bit key. Is this security configuration considered deprecated?

I am still wondering why the server's certificate and not the client one's is being revoked when establishing a connection...

mregen, 2018-04-17:
Hi @Matthumba, SHA1 signed certs are considered deprecated. The client may not accept the SHA256 signed certs? Not recommended, but you try to delete the old server cert and default the key and hash size for the server by modifying:
```
public class CertificateFactory
    public const ushort defaultKeySize = 2048;  --> 1024
    public const ushort defaultHashSize = 256;  --> 160
 ```

and see if it works.

Matthumba, 2018-04-27:
Finally, I was able to solve the problem. I have to import the server's certificate to the local machine certificate store. The Windows certificate import wizard only uses the store for current user.

But thanks anyway!

closed 2018-04-27 by Matthumba

NakWarsi, 2019-02-04:
@Matthumba  I am facing exactly same problem can you let me know how di you do that ?

cacamille3, 2019-02-04:
On Windows, double click on Certificate, Choose `Install Certificate...`
Then select `Local Machine `
Then `Place all certificates in the following store` and _Browse_ to **Trusted Root Certification Authorities.**

NakWarsi, 2019-02-05:
@cacamille3 
got it,
thanks