OPCFoundation/UA-.NETStandard #427: How to create and manage Application Certififcate

msalafia, 2018-04-14:
I'm trying to create certificate with openssl and save these certificate in a folder.
i have created my certificates with the following commands 

`openssl req -new -sha256 -key ./private/webplatform.cert.pem -subj "/CN=OPCUAWebPlatform/DC=dlserver1" -out webplatform.csr.der`

`openssl ca -config openssl.cnf -days 375 -notext -md sha256 -in webplatform.csr.der -out webplatform.cert.der`

So i put this certificate in a folder local to the client application project folder with the relative path "./OPC Foundation/CertificateStores/MachineDefault".

I load the app configuration as the following:

```csharp
private ApplicationConfiguration CreateAppConfiguration(string applicationName, int sessionTimeout)
        {
            var config = new ApplicationConfiguration()
            {
                ApplicationName = applicationName,
                ApplicationType = ApplicationType.Client,
                ApplicationUri = "urn:localhost:OPCFoundation:" + applicationName,
                SecurityConfiguration = new SecurityConfiguration
                {
                    ApplicationCertificate = new CertificateIdentifier
                    {
                        StoreType = "Directory",
                        StorePath = "./OPC Foundation/CertificateStores/MachineDefault",
                        SubjectName = Utils.Format("CN={0}, DC={1}", applicationName, Utils.GetHostName())
                    },
                    TrustedPeerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "./OPC Foundation/CertificateStores/UA Applications",
                    },
                    TrustedIssuerCertificates = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "./OPC Foundation/CertificateStores/UA Certificate Authorities",
                    },
                    RejectedCertificateStore = new CertificateTrustList
                    {
                        StoreType = "Directory",
                        StorePath = "./OPC Foundation/CertificateStores/RejectedCertificates",
                    },
                    NonceLength = 32,
                    AutoAcceptUntrustedCertificates = true
                },
                TransportConfigurations = new TransportConfigurationCollection(),
                TransportQuotas = new TransportQuotas { OperationTimeout = 15000 },
                ClientConfiguration = new ClientConfiguration { DefaultSessionTimeout = sessionTimeout }
            };

            return config;
        }
```
 But when i call the `Session.Create` method it throws an exception because it cannot find the certificate. This is the point in the `Session.Create` method:

```csharp
clientCertificate = await configuration.SecurityConfiguration.ApplicationCertificate.Find(true);

                if (clientCertificate == null)
                {
                    throw ServiceResultException.Create(StatusCodes.BadConfigurationError, "ApplicationCertificate cannot be found.");
                }
```
 I dived with the debugger into all the methods calls an i found out that the problem is due to the following piece of code of CertificateIdentifier.Find method

```csharp
// find by subject name.
            if (!String.IsNullOrEmpty(subjectName))
            {
                List<string> subjectName2 = Utils.ParseDistinguishedName(subjectName);

                foreach (X509Certificate2 certificate in collection)
                {
                    if (Utils.CompareDistinguishedName(certificate, subjectName2))
                    {
                        if (!needPrivateKey || certificate.HasPrivateKey) //<----- needPrivateKey=true, certificate.HasPrivateKey=false
                        {
                            return certificate;
                        }
                    }
                }

                collection = collection.Find(X509FindType.FindBySubjectName, subjectName, false);

                foreach (X509Certificate2 certificate in collection)
                {
                    if (!needPrivateKey || certificate.HasPrivateKey) //<----- needPrivateKey=true, certificate.HasPrivateKey=false
                    {
                        return certificate;
                    }
                }
            }
// certificate not found.
            return null;     //<----- HERE THE NULL CERTIFICATE
```

So the problem seems to be the certificate.HasPrivateKey = false. So i tried to understand why it is set in this way but i was not able to do that. The only thing i found is that in the method `private IDictionary<string, Entry> Load(string thumbprint)` of DirectoryCertificateStore, the entry in the returned Dictionary containing my certificate correctly shows the path to the relevant private key. But unfortunately the certicate instance in the entry (class X509Certificate2) presents a property HasPrivateKey = false.

So is there a problem in the way i created the certificate perhaps? Please i don't know how to handle this problem using a DirectoryCertificateStore, because the sample application you provide use X509CertificateStore.

dougjr17, 2018-04-15:
Pretty sure you need to combine the original key "./private/webplatform.cert.pem" with the signed certificate "webplatform.cert.der"

Check out this link for how to convert your DER file to PEM, then merge it with the original PEM key file and output a PFX/PKCS12 or whatever it may be called where public and private keys are in same certificate file: https://github.com/node-opcua/node-opcua/blob/master/documentation/notes_on_certificates.md

Hope this helps.

msalafia, 2018-04-16:
Thank you very much for your help! So, as i understand there is no way to use separate files for both the application certificate and the relevant private key, is it correct?

dougjr17, 2018-04-16:
There is a way to supply the key from separate file in .NET (see [this post](https://stackoverflow.com/questions/18462064/associate-a-private-key-with-the-x509certificate2-class-in-net)); however, in the original post you stated that:

> in the method private IDictionary<string, Entry> Load(string thumbprint) of DirectoryCertificateStore, the entry in the returned Dictionary containing my certificate correctly shows the path to the relevant private key 

So my recommendation would be to just run:
`openssl pkcs12 -export -in webplatform.cert.der -inkey ./private/webplatform.cert.pem -out webplatform.cert.pfx` and use this PKCS12 ".pfx" file instead. 

If all else fails, you can either build the "OPC UA .NET Certificate Generation Tool" [from here](https://github.com/OPCFoundation/Misc-Tools), however that in itself has challenges and caveats described in that [project's issues found here](https://github.com/OPCFoundation/Misc-Tools/issues/5).

If you are an OPC Foundation Member, you may also check out their ["OPC UA Sample Applications"](https://opcfoundation.org/developer-tools/developer-kits-unified-architecture/sample-applications/) - unfortunately I cannot comment further due to the [OPC Foundation's licensing agreement found here.](https://opcfoundation.org/license/binaries/1.00/index.html)

msalafia, 2018-04-16:
I solved using the solutions proposed in the Sample Application in this repo. Thank you anyway for clarifying all my doubts. Very kind of you.

closed 2018-04-16 by msalafia

jangrygerek, 2021-06-16:
Hallo, please how you solved this issue? I can not find solution. Thank you.

> I solved using the solutions proposed in the Sample Application in this repo. Thank you anyway for clarifying all my doubts. Very kind of you.

rafaelschlatter, 2021-07-23:
Would also be interested which sample app you are referring to @msalafia , and how you actually solved the issue.