OPCFoundation/UA-.NETStandard #429: Publish the OPCUA .net core web application  project on Ubuntu 16.04

vrushalikatkade, 2018-04-17:
I created the .net core web application . to access the simple list of OPCUA server on Ubuntu 16.04 .after publishing the web application it throws the following exception 

*******************************************
An unhandled exception occurred while processing the request.
NullReferenceException: Object reference not set to an instance of an object.
OPCUABrowse.Pages.OPCUABrowse_Page+<<ExecuteAsync>b__8_0>d.MoveNext()

NullReferenceException: Object reference not set to an instance of an object.
OPCUABrowse.Pages.OPCUABrowse_Page+<<ExecuteAsync>b__8_0>d.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Razor.Runtime.TagHelpers.TagHelperExecutionContext+<SetOutputContentAsync>d__30.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
OPCUABrowse.Pages.OPCUABrowse_Page+<ExecuteAsync>d__8.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.Razor.RazorView+<RenderPageCoreAsync>d__16.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.Razor.RazorView+<RenderPageAsync>d__15.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
System.Runtime.CompilerServices.TaskAwaiter.GetResult()
Microsoft.AspNetCore.Mvc.Razor.RazorView+<RenderAsync>d__14.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.ViewFeatures.ViewExecutor+<ExecuteAsync>d__22.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker+<InvokeResultAsync>d__19.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker+<InvokeNextResultFilterAsync>d__24.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker.Rethrow(ResultExecutedContext context)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker.Next(ref State next, ref Scope scope, ref object state, ref bool isCompleted)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker+<InvokeNextResourceFilter>d__22.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker.Rethrow(ResourceExecutedContext context)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker.Next(ref State next, ref Scope scope, ref object state, ref bool isCompleted)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker+<InvokeFilterPipelineAsync>d__17.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Mvc.Internal.ResourceInvoker+<InvokeAsync>d__15.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Builder.RouterMiddleware+<Invoke>d__4.MoveNext()
System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
Microsoft.AspNetCore.Diagnostics.DeveloperExceptionPageMiddleware+<Invoke>d__7.MoveNext()
*******************************************************************

I have used the OPCUA nuget pakage  https://www.nuget.org/packages/OPCFoundation.NetStandard.Opc.Ua/

my application code file 

[my applicaion code.txt](https://github.com/OPCFoundation/UA-.NETStandard/files/1919766/my.applicaion.code.txt)

my chtml page code 

@page
@model OPCUABrowseModel
@using Opc.Ua;
<html>
<body>
      <label for="gender1" class="col-sm-2 control-label"> Host Name:</label><input id="Text1" type="text" /> <input id="Button1" type="button" value="Discover" />
    <br />
    <div class="form-group">
        <label for="gender1" class="col-md-2 control-label" >Server Name :</label>
        <div class="col-md-10">
            <select class="form-control" id="gender1">
                @foreach (ApplicationDescription Servername in Model.servers)
                {
                    <option>@Servername.ApplicationName</option>
                }
            </select>
        </div>
        <br />
        <br />
    </div>
</body>
</html>

Thanks ,
vrushali

vrushalikatkade, 2018-04-20:
please suggest any solution

AlinMoldovean, 2018-04-24:
Hello @vrushalikatkade ,

From the call stack it seems that NullReferenceException exception is thrown in the source code of your application (OPCUABrowse.Pages).

You need to make a debug and get the exact source of the problem.

closed 2018-06-14 by AlinMoldovean