OPCFoundation/UA-.NETStandard #430: Documentation - heads & tails

zlatinejc, 2018-04-19:
I would really like to see some documentation for newcomers making some heads and tails with this great repository. 

Although this repository is very rich in components and solutions, it is very hard to make any sense where to start in regard to:

* understanding how components work in general (LDS, REFERENCE server, SERVER, CLIENT)
* setting up a simple client server authentication (local & remote)
* running sample servers using in domain like environment
* how to use UA configuration tool for certificate management
* any other general use cases
* understanding common pitfalls & best practices

I think this would really be a bar raiser in general understanding of this magnificent repo.

goyzhang, 2018-05-21:
It makes me feel that the ReferenceServer/Client are heavily source-code-based. It is hard to get started since overrides are everywhere instead of interfaces&methods. Since there is no clear documentation, I have to trace back layer-by-layer trying to understand how it works.

AlinMoldovean, 2018-06-14:
For the moment there is no detailed help file for this repository.
You can get some better documentation from commercial SDK vendors like Softing, Unified Automation.

alexbeeston, 2019-06-10:
I am writing an OPC Server as well, and I completely agree with the above comments. I'm so glad this starter code is available, but it's laborious to walk through it and try to make sense of what's going on. @caacree, did you finish the "walk-through" you mentioned, and if so, where could I find it?

For the authors of the repo, there is still a need for documentation at least  a year after this discussion thread was created.

MichaelGehling, 2019-07-04:
I would also be greatful for some Reference Documentation with the API explained in detail and maybe a step by step tutorial, though the sample projects definetly help!

mpostol, 2019-07-06:
I am working on a new concept called Object-Oriented Internet that derives from OPC UA and reuse the standard. An important part of this project is an ebook - that I thought - covers many topics related to OPC UA. It is available here:

[Object-Oriented Internet ebook](https://commsvr.gitbook.io/ooi/readme)

Let me know what is missing and I will try to harmonize my priorities with your needs. You may provide me feedback creating a new issue on [GitHub](https://github.com/mpostol/OPC-UA-OOI) or drop a few words on [Gitter](https://gitter.im/mpostol/OPC-UA-OOI).

Let's work together, but keep in mind that the OPC UA Spec is just a paper containing a couple of thousands of pages covering awesome ideas and tremendous errors.  Everything developed on top of it must hide details but preserve the main idea relevant for different users, e.g. developers, integrators, end users, architects, decision makers, investors, etc. It is really difficult to find a trade-off. Provided by the OPC UA product vendors documentation, i.e. SDK manuals and product user guides are only examples partially covering all the needs. 

> I proposed it many times so let me recall it once more.
>
> **I believe that education should be the OPC Foundation responsibility planned and fulfilled methodically.**

I am also sure that this activity will support also the OPC UA product vendors, i.e. it must not be recognized as an activity against them for sure. Position on the marked not always is a good cause to be a robust education provider. 

> Even if the education is serviced by the independent organization it must be harmonized, regulated, validated or even certified by OPC Foundation.

OPC UA is common good so OPC Foundation must take care about how to use it.

noumanqaiser, 2019-10-09:
Hi Team,
I am a beginner and wanted to use this library to simply read data (ua .net client) from a Kepware KepServerEx OPC UA server. Honestly I struggled a lot but eventually managed to write code that does that.

I am more then happy to create a tutorial that allows people with basic knowledge of .NET  to create a simple application that allows them to read tags from OPC UA server. Any views on how where this tutorial would be hosted?

VidyaPuri, 2019-12-03:
> Hi Team,
> I am a beginner and wanted to use this library to simply read data (ua .net client) from a Kepware KepServerEx OPC UA server. Honestly I struggled a lot but eventually managed to write code [...]

Did you manage to create the mentioned tutorial? I am at the place where i am struggling to get to the data sent by server. I am using the console client sample, get it to connect, but i am unable to subscribe or to get the desired MonitoredItem value to be writen out with Notification method.

noumanqaiser, 2019-12-03:
Hi @VidyaPuri 
I have the code ready. Can you post this as a question on Stackoverflow and I will write a detailed reply possibly today, that way I think it will help the maximum number of people.

Regards,
Nouman

VidyaPuri, 2019-12-04:
Hi @noumanqaiser 

i have made a post on Stackoverflow as you requested! its ere
[Stackoverflow post](https://stackoverflow.com/questions/59171684/opc-ua-net-standard-c-sharp-simple-console-client)
Thank you very much,
Vid

noumanqaiser, 2019-12-04:
Hi Vidya, 
I saw that your question is already closed on stackoverflow by some moderators, can you edit and open it again.

noumanqaiser, 2019-12-04:
Hi Vidya,
I will post the solution on stackoverflow once your question is open aswell. here is the answer for your reference.

I struggled with very similar problems with the fact that many online OPC UA examples including the ones on OPC UA foundation are not simple to be deployed. 

I tried the solution below to read values of tags from kepware ServerEX OPC UA Server and works stable, I hope same would probably work for Prosys aswell. 

Ensure you install the Nuget Package: OPCFoundation.NetStandard.Opc.Ua

`
using System;
using System.Collections.Generic;
using System.Windows.Forms;

using Opc.Ua;   // Install-Package OPCFoundation.NetStandard.Opc.Ua
using Opc.Ua.Client;
using Opc.Ua.Configuration;

using System.Threading;

namespace Test_OPC_UA
{
    public partial class Form1 : Form
    {
        //creating a object that encapsulates the netire OPC UA Server related work
        OPCUAClass myOPCUAServer;

        //creating a dictionary of Tags that would be captured from the OPC UA Server
        Dictionary<String, Form1.OPCUAClass.TagClass> TagList = new Dictionary<String, Form1.OPCUAClass.TagClass>();

        public Form1()
        {
            InitializeComponent();
           

            //Add tags to the Tag List, For each tag, you have to define the name of the tag and its address
            //the address can typically be found by browsing the OPC UA Server's tree. In the example below
            // The OPC Server had the following hierarchy: M0401 -> CPU945 -> IBatchOutput
            //i used TBC0401 as a name of the tag, you can use any name
            //add as many tags as you want to capture
            TagList.Add("TBC0401", new Form1.OPCUAClass.TagClass("TBC0401", "M0401.CPU945.iBatchOutput"));

            //to initialize the OPC UA Server, provide the IP Address, Port Number, the list of tags you want to capture
            //in some OPC UA servers and kepware aswell the session can be closed by the OPC UA Server, so its better to 
            //allow the class to reinitiate session periodically, before renewing current sessions are closed
            myOPCUAServer = new OPCUAClass("127.0.0.1", "49320", TagList, true, 1, "2");

            //once the OPC Server has been initialized, you can easily read Tag values and even see when they were
            // updated last time
            //as an example i could read the TBC0401 tag by:

            var tagCurrentValue = TagList["TBC0401"].CurrentValue;
            var tagLastGoodValue = TagList["TBC0401"].LastGoodValue;
            var lastTimeTagupdated = TagList["TBC0401"].LastUpdatedTime;

        }

        public class OPCUAClass
        {
            public string ServerAddress { get; set; }
            public string ServerPortNumber { get; set; }
            public bool SecurityEnabled { get; set; }
            public string MyApplicationName { get; set; }
            public Session OPCSession { get; set; }
            public string OPCNameSpace { get; set; }
            public Dictionary<string, TagClass> TagList { get; set; }

            public bool SessionRenewalRequired { get; set; }
            public double SessionRenewalPeriodMins { get; set; }
            public DateTime LastTimeSessionRenewed { get; set; }
[... 189 lines cut ...]
                                TagList[item.DisplayName].LastSourceTimeStamp = value.SourceTimestamp.ToLocalTime();
                                TagList[item.DisplayName].StatusCode = value.StatusCode.ToString();

                            }
                            else
                            {
                                TagList[item.DisplayName].StatusCode = value.StatusCode.ToString();
                                TagList[item.DisplayName].CurrentValue = null;

                            }

                        }

                    }

                }
                InitialisationCompleted = true;
            }

        }
        
    }
}

`

VidyaPuri, 2019-12-04:
Thank you very much, i will try to incorporate your solution into mine and see how it works. Btw i have edited the stackoverflow post, so now it is open for answers! 

regards,
Vid

noumanqaiser, 2019-12-04:
Great. Do try and let us know how if you face any issues. I think your principal problem of extracting value of the tag should be resolved.

noumanqaiser, 2019-12-30:
HI @VidyaPuri , Did the solution work for you? and if yes, can you mark it as answered in stackoverflow

Nouman

VidyaPuri, 2020-01-01:
I was on vacation for a month sorry for delay! I have marked it answered. Btw are you still working with this librarry? Would love to "stay in touch" for exchanging ideas and issues and solutions :)

closed 2020-11-06 by AlinMoldovean