OPCFoundation/UA-.NETStandard #431: Could not verify security on OpenSecureChannel request

zlatinejc, 2018-04-19:
**ISSUE**

Cannot connect to remote UA sample server: 
opc.tcp://srvtistk01.ourdomain.com:62547/Quickstarts/DataAccessServer

**USING**

```
dotnet run --project NetCoreConsoleClient.csproj -a opc.tcp://srvtistk01.ourdomain.com:62547/Quickstarts/DataAccessServer
```

SampleClient.Config.xml
```xml
<AutoAcceptUntrustedCertificates>true</AutoAcceptUntrustedCertificates>
<RejectSHA1SignedCertificates>false</RejectSHA1SignedCertificates> 
<MinimumCertificateKeySize>1024</MinimumCertificateKeySize>
```

**PROBLEM**

I am using the OPC UA DataAccessServer from the opcfoundation.org. The server url is running on localhost. Although I can connect to OPC UA GenericServer sample remotely I do not seem to posses the knowledge to configure the DataAccessServer to accept my client. Any help would be appreciated.

**DEBUG LOG:**

Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Private.CoreLib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/projekti/OPCFoundation/UA-.NETStandard/SampleApplications/Samples/NetCoreConsoleClient/bin/Debug/netcoreapp2.0/NetCoreConsoleClient.dll'. Symbols loaded.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/.nuget/packages/mono.options/5.3.0.1/lib/netstandard1.3/Mono.Options.dll'. Module was built without symbols.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.ObjectModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/projekti/OPCFoundation/UA-.NETStandard/SampleApplications/Samples/NetCoreConsoleClient/bin/Debug/netcoreapp2.0/Opc.Ua.Core.dll'. Symbols loaded.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/netstandard.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Console.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Collections.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.IO.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Step into: Stepping over non-user code 'NetCoreConsoleClient.Program.<>c__DisplayClass0_0..ctor'
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.IO.FileSystem.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.InteropServices.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Threading.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Text.Encoding.Extensions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
.Net Core OPC UA Console Client sample
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Text.RegularExpressions.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/projekti/OPCFoundation/UA-.NETStandard/SampleApplications/Samples/NetCoreConsoleClient/bin/Debug/netcoreapp2.0/Opc.Ua.Client.dll'. Symbols loaded.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Threading.Tasks.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/projekti/OPCFoundation/UA-.NETStandard/SampleApplications/Samples/NetCoreConsoleClient/bin/Debug/netcoreapp2.0/Opc.Ua.Configuration.dll'. Symbols loaded.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Security.Cryptography.X509Certificates.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/.nuget/packages/system.servicemodel.primitives/4.3.0/lib/netstandard1.3/System.ServiceModel.Primitives.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/Users/nejctrnjanin/.nuget/packages/system.private.servicemodel/4.3.0/runtimes/unix/lib/netstandard1.3/System.Private.ServiceModel.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
1 - Create an Application Configuration.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Diagnostics.Debug.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
04/19/2018 10:24:40.462 Loading application configuration file. /Users/nejctrnjanin/projekti/OPCFoundation/UA-.NETStandard/SampleApplications/Samples/NetCoreConsoleClient/Opc.Ua.SampleClient.Config.xml
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.Serialization.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Private.DataContractSerialization.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Xml.ReaderWriter.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Private.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Private.Uri.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.Serialization.Primitives.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Resources.ResourceManager.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Linq.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Reflection.Emit.ILGeneration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Reflection.Emit.Lightweight.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Reflection.Primitives.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Runtime.Serialization.Formatters.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Diagnostics.Tools.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Loaded '/usr/local/share/dotnet/shared/Microsoft.NETCore.App/2.0.5/System.Collections.NonGeneric.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.

************************* Logging started at 04/19/2018 10:24:40
[... cut ...]
Exception thrown: 'System.AggregateException' in System.Private.CoreLib.dll
04/19/2018 10:24:44.068 ServiceResultException:One or more errors occurred. (Error establishing a connection: Error received from remote host: Could not verify security on OpenSecureChannel request.)
Exception: One or more errors occurred. (Error establishing a connection: Error received from remote host: Could not verify security on OpenSecureChannel request.)
The program '[30224] NetCoreConsoleClient.dll' has exited with code 0 (0x0).

markkuhar, 2018-04-19:
+1 I am facing the same issue.

AlinMoldovean, 2018-04-19:
Hello @zlatinejc  , @markkuhar ,

The connection seems to be rejected from the server side.
Are you attempting to connect on an endpoint with or without security?
The client certificate was copied in the trust list of the server?

zlatinejc, 2018-04-19:
Hi!

The DataAccess client that I am using on the server machine (just for testing) lists some URLs among which I use the unsecured without security (at least I think so).

The Cert was copied, to the store via the UA Configuration manager. Is there a physical localtion I can check if it is actually there ?

Thank you for your help.

AlinMoldovean, 2018-04-19:
@zlatinejc ,

The location of the trusted certificates should be in the configuration of the server.
That is the place where you need to copy the public key of the client.

zlatinejc, 2018-04-19:
It seems that every time I try to establish a connection to the server my certificate move to RejectedCertificates folder:

C:\ProgramData\OPC Foundation\CertificateStores\RejectedCertificates\certs

Any ideas how to fix this ?

zlatinejc, 2018-04-19:
Solved:

Needed to import the cert to **%CommonApplicationData%\OPC Foundation\CertificateStores\UA Applications** store using the **Import Certificate To Store...**

[image]

It worked.

Thank you for your help.

markkuhar, 2018-04-19:
Thank you very much!

closed 2018-04-19 by AlinMoldovean

inmen, 2020-05-21:
application.ApplicationConfiguration.CertificateValidator.CertificateValidation += CertificateValidator_CertificateValidation;

`        private void CertificateValidator_CertificateValidation(CertificateValidator sender, CertificateValidationEventArgs e)
        {
            e.Accept = true;
            var store = new X509Store(StoreName.Root, StoreLocation.CurrentUser);
            store.Open(OpenFlags.ReadWrite);
            store.Add(e.Certificate);
            store.Close();
        }`