OPCFoundation/UA-.NETStandard #433: Brief explanation of browse functionality needed

zlatinejc, 2018-04-19:
Can someone explain how does OPC browsing nodes work ?

As far as I understand it there are 2 types of Nodes (references & non-references).

I am struggling to understand what is the diff between these 2 types of browsing methods:

**1. METHOD**
```java
public ReferenceDescriptionCollection FetchReferences(NodeId nodeId)
        {
            // browse for all references.
            byte[] continuationPoint;
            ReferenceDescriptionCollection descriptions;

            Browse(
                null,
                null,
                nodeId,
                0,
                BrowseDirection.Both,
                null,
                true,
                0,
                out continuationPoint,
                out descriptions);

            // process any continuation point.
            while (continuationPoint != null)
            {
                byte[] revisedContinuationPoint;
                ReferenceDescriptionCollection additionalDescriptions;

                BrowseNext(
                    null,
                    false,
                    continuationPoint,
                    out revisedContinuationPoint,
                    out additionalDescriptions);

                continuationPoint = revisedContinuationPoint;

                descriptions.AddRange(additionalDescriptions);
            }

            return descriptions;
        }
```

**2. METHOD**

```java
 session.Browse(
                null,
                null,
                ObjectIds.ObjectsFolder,
                0u,
                BrowseDirection.Forward,
                ReferenceTypeIds.HierarchicalReferences,
                true,
                (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
                out continuationPoint,
                out references);

            Console.WriteLine(" DisplayName, BrowseName, NodeClass");
            foreach (var rd in references)
            {
                Console.WriteLine(" {0}, {1}, {2}", rd.DisplayName, rd.BrowseName, rd.NodeClass);
                ReferenceDescriptionCollection nextRefs;
                byte[] nextCp;
                session.Browse(
                    null,
                    null,
                    ExpandedNodeId.ToNodeId(rd.NodeId, session.NamespaceUris),
                    0u,
                    BrowseDirection.Forward,
                    ReferenceTypeIds.HierarchicalReferences,
                    true,
                    (uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
                    out nextCp,
                    out nextRefs);

                foreach (var nextRd in nextRefs)
                {
                    Console.WriteLine("   + {0}, {1}, {2}", nextRd.DisplayName, nextRd.BrowseName, nextRd.NodeClass);
                }
            }
)
```

My general concerns are:

* what is the diff between continuation point and the reference point 
* what is the purpose of fetching references, meaning why not fetch all points recursively 
* is there any performance / best practice pitfall
* what is the meaning of ``` BrowseDirection``` & ```continuation points``` and why are they needed here in particular

Thank you for your patience.

zlatinejc, 2018-04-19:
My guess:

* BrowseDirection : means traversing the three up or down ?
* continuation points : could this mean some kind of paginaton ?
* Can a server force a default pagination per se ?

AlinMoldovean, 2018-05-17:
Hi @zlatinejc ,

The question is related to the Browse service specified in the Part 4. of the OPC UA Specification.
The document contains a detailed description of all services, including Browse.

closed 2018-05-17 by AlinMoldovean

zlatinejc, 2018-05-23:
Is there a link I can follow up ?

AlinMoldovean, 2018-05-25:
The OPC UA Specification documents can be found here.
https://opcfoundation.org/developer-tools/specifications-unified-architecture
Some of them are available only for OPC Foundation members.