OPCFoundation/UA-.NETStandard #435: How to display Folders, etc

NichtEuler, 2018-04-26:
Hello,

I have created a Address Space Model with:
http://www.commsvr.com/tabid/533/language/en-US/Default.aspx
and compiled the output with the UA-ModelCompiler-Tool.
The Outputs are [Demo.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/1950562/Demo.zip).
Basically its a Folder organized by Objects. This folder is containing another folder with a variable.

I added the line: 
nodeManagers.Add(new global::Demo.DemoNodeManager(server, configuration));
to Opc.Ua.Sample.SampleServer.CreateMasterNodeManager.

So here is my question: How can I set up the DemoNodeManager in a way, I am able to see these folders I created?

I looked up the other NodeManagers for reference, but couldn't figure it out.

Help would be much appreciated.

(P.S.: I am studying electrical engineering in my 4.th term, so I am no skilled information scientist. Maybe a sample would be nice to have :) )

AlinMoldovean, 2018-05-17:
Hi @NichtEuler ,

Please take a look at the ReferenceServer project.
It contains code for adding different kind of nodes, including Folders.

NichtEuler, 2018-06-04:
Oh, I did not see that.

closed 2018-06-04 by NichtEuler