OPCFoundation/UA-.NETStandard #436: Executing a method via the MethodType and the ObjectId

Noires, 2018-04-27:
According to the OPC UA Specification (1.04.11) Part 4 around Page 57-58 it is defined that you can use the type of the method with the parent object of the method to execute the method, which is not possible at the moment via the SampleNodeManager. Since there is a check (at line 1811) which says that the method is invalid if it does not belong directly to the object:

```
                    // check if method belongs to the object.
                    if (!Object.ReferenceEquals(method.Parent, source))
                    {
                        errors[ii] = StatusCodes.BadMethodInvalid;
                        continue;
                    }
```

Is there a reason for this check? We had a discussion at a meeting for a specification for opc ua and  came to the conclusion that it should be possible, but maybe you could clear things up.

AlinMoldovean, 2018-06-15:
Hi @Noires ,

I seems that according to the specifications the call should be allowed in case there is an "HasComponent" reference between ObjectId node and MethodId.

AlinMoldovean, 2018-06-19:
Validation in SampleNodeManager now accepts also ObjectId with HAsComponent reference to MethodsId

commit 1c7a208d (2018-06-19): Fixes #436. Update method call validation in SampleNodeManager. (#477)
closed 2018-06-19 by AlinMoldovean