OPCFoundation/UA-.NETStandard #442: Server Push Management - Exception with UA Ansi C Server on "Request New" Certificate

rbeex, 2018-05-06:
When using the UA Ansi C Server with GDS Client "Server Push Configuration". An exception occurs on request a new certificate with "Request New".

Fix:
In "ApplicationCertificateControl.cs" this lines
   byte[] certificateRequest = m_server.CreateSigningRequest(
       null,
       m_server.ApplicationCertificateType,

change to 

   byte[] certificateRequest = m_server.CreateSigningRequest(
       m_server.DefaultApplicationGroup,
       m_server.ApplicationCertificateType,

and this lines
  bool applyChanges = m_server.UpdateCertificate(
       null,
       m_server.ApplicationCertificateType,

change to 

  bool applyChanges = m_server.UpdateCertificate(
       m_server.DefaultApplicationGroup,
       m_server.ApplicationCertificateType,

See http://documentation.unified-automation.com/uasdkcpp/1.5.2/html/classUaClientSdk_1_1UaServerConfigurationObject.html#a114c225fbe48ee203069aa3f34380ed7

[in] | certificateGroupId | The certificateGroupId parameter allows the caller to specify a Certificate Group that provides context for the request. If null the CertificateManager shall choose DefaultApplicationGroup.

Maybe the bug is in the UA Ansi C Server. But with this fix, it does work.

mregen, 2018-05-07:
Hi @rbeex, I believe this issue has been fixed in latest Unified Automation Server SDK (C and C++) releases, around Dec. 2017. Please check with latest SDK. Previously the UA SDK implementation did not follow the offical spec which allows for null parameters.

rbeex, 2018-05-07:
Thanks for the info @mregen . I used the last "prebuild" UA ANSI C Server from Unified Automation, downloaded a few days ago. Meybe this is an older one.
Don´t wonder who is send you an request on linkedin. I´m these guy.

mregen, 2018-05-08:
@rbeex c++ or C? I *think* the UA Ansi C server should behave compliant, but you better follow up with Unified Automation on the issue. 
We don't want to establish a 'de facto' standard by always passing the default parameters, because null parms are spec compliant.

closed 2018-05-25 by AlinMoldovean