OPCFoundation/UA-.NETStandard #444: Getting the ExpandedNodeIds of the Nodes in the TypeDictionary for the Client

Noires, 2018-05-17:
Hello together,

iam currently implementing the handling of complex data on the client side (like mentioned in #243).

The reading of the type dictionary works and even creating the types at runtime based on a generic abstract type class with the Interface IEncodeable. The problem is that i dont know where to get the ExpandedNodeId to add the types properly to the EncodeableFactory. 

As a example during a read request i get the ExpandedNodeId for the Binary Decoding of a Type and can get with that the ExpandedNodeId for the DataType, but i rather would have it at the start of a connection so i can build up the whole dictionary and all types beforehand.

My current steps:
1. Connecting to server and reading the binary data for the typed dictionaries.
2. Converting the binary data and parsing the typed dictionaries.
3. Creating the types via runtime with Refleciton and Emit based on classes like the following and adding properties and overwriting NodeId and Namespace:
`    public abstract class GenericEncodeableNodeClass:IEncodeable
    {
        public void Encode(IEncoder encoder)
        {
            // Encoding via Reflection
        }

        public void Decode(IDecoder decoder)
        {
            // Decoding via Reflection
        }

        public bool IsEqual(IEncodeable encodeable)
        {
           // IsEqual via Reflection
        }

        public abstract uint NodeId { get; }
        public abstract string Namespace { get; }

        public ExpandedNodeId TypeId => new ExpandedNodeId(NodeId, Namespace);
        public ExpandedNodeId BinaryEncodingId => new ExpandedNodeId(NodeId, Namespace);
        public ExpandedNodeId XmlEncodingId => new ExpandedNodeId(NodeId, Namespace);
    }` 
4. Adding the created dynamic assemblies to the encodeable factory.

Basically at step 3 i need the NodeId.

Any idea where to get it? I rather would have not to do things like read the whole address space and match things myself.

AlinMoldovean, 2018-05-22:
Hello @Noires ,

You need to perform the following steps:
 - Browse the Dictionary node inside the "OPC Binary" node for HasComponent references. It should return the list of structured types from the dictionary
 - For each result, first make an Inverse browse for HasDescription references. This should return the "BinaryEncodingId" of the type
 - For the result, make an Inverse HasEncoding browse. This should return the TypeId of the structured type.
 - Optionally for XML encoding support, make a Forward HasEncoding browse on the TypeId. It should return XmlEncodingId with BrowseName = "Default XML", and BinaryEncodingId with BrowseName + "Default Binary"

[image]

Noires, 2018-05-25:
Thanks for the detailed answer!

The dynamic types are being properly generated now and used by the binary decoder, so everything is working as it should be.

closed 2018-05-25 by Noires

Noires, 2018-08-07:
@AlinMoldovean 
It is nearly working as expected. Sadly if i got a complex type which consists of a custom enum, it doesnt work. The only suitable solution at the moment is browsing the whole addressspace.  Do you have any idea how i should proceed there?

As a example: I can get the NodeId for the type ProductionInformation, which has a enum property of the type ProductionInformationEnum. How do i get the informations for the enum? I dont find any references.

AlinMoldovean, 2018-08-07:
@Noires ,

In some cases the server provides the information for the enum fields in the type dictionary.
e.g.
```
<opc:TypeDictionary
  xmlns:opc="http://opcfoundation.org/BinarySchema/"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:ua="http://opcfoundation.org/UA/"
  xmlns:tns="http://industrial.softing.com/UA/Refrigerator"
  DefaultByteOrder="LittleEndian"
  TargetNamespace="http://industrial.softing.com/UA/Refrigerator"
>
  <!-- This File was generated on 2014-06-02 and supports the specifications supported by version 1.25.0.3534 of the OPC UA deliverables. -->
  <opc:Import Namespace="http://opcfoundation.org/UA/" Location="Softing.Opc.Ua.Sdk.BinarySchema.bsd"/>

  <opc:StructuredType Name="RefrigeratorStatusType" BaseType="ua:ExtensionObject">
    <opc:Field Name="CondensorMotorRunning" TypeName="opc:Boolean" />
    <opc:Field Name="PreasureBeforePump" TypeName="opc:Double" />
    <opc:Field Name="PreasureAfterPump" TypeName="opc:Double" />
  </opc:StructuredType>

  <opc:EnumeratedType Name="EnumRefrigeratorState" LengthInBits="32">
    <opc:EnumeratedValue Name="Stopped" Value="0" />
    <opc:EnumeratedValue Name="Running" Value="1" />
    <opc:EnumeratedValue Name="Fault" Value="2" />
  </opc:EnumeratedType>

</opc:TypeDictionary>
```
The remaining part is to browse the "Enumeration" node (ns=0;i=29) for HasSubtype and find the reference with the same BrowseName as the one from dictionary. The NodeId of that node is the TypeId of the enumerated type.

In other cases the enumeration type are only exposed bellow "Enumeration" node (ns=0;i=29) .
In that case the client needs to read the _EnumValues_ or/and _EnumStrings_ properties of the custom enumeration types in order to load enum type information.

Noires, 2018-08-07:
Thank you so much for the quick answer. Got it working now and its quick and smooth. I will try to get a sample online, when i refactored and cleaned up the code. I just wasn't expecting the Enumerations below the Enumeration Node, which actually makes sense though.

davidemess, 2018-10-01:
Hi @Noires,
Have you worked on a sample? I'm trying to read structured data types from a PLC that work as server. I don't need a dynamically association of DataType. I would create a class (interface) on the client that worked as dataType and pass it when I Read a determinated structure on the Server. Can you help me?

Abarbz, 2018-10-12:
@davidemess here are some samples that I hope will help you: 
https://gist.github.com/Abarbz/459fddc19c718568bbc39499251dbbc2
In my case, the PLC is from Beckhoff, so you will probably have to adapt the `ExpandedNodeId.Parse(
				"nsu=urn:BeckhoffAutomation:Ua:PLC1;s=<StructuredDataType>:ComplexStruct");` part. 
To help you, here is how I figured out those values using debug mode in VS: 
[image]

davidemess, 2018-10-12:
Hi @Abarbz,
Thank you for your support! Currently I read Structs like Byte [] and I create methods for decode them. I use Omron Plc, I asked to Omron staff about Opcua and  I have understand that they don't use a single node for each propriety into the Struct so I can't do "ns = 4, s = MAIN.complexStruct.Variable".

 I think your code will help me about the Interface EncodeableObject that I did not understand so much, and I don't know if i can implement it with Omron.

Thank you!

mregen, 2019-10-31:
see als #838