OPCFoundation/UA-.NETStandard #450: Why the 'PolicyId' in 'UserTokenPolicy' always 'null'

diverger, 2018-05-20:
When I run the 'SampleServer', and step into 'CreateUserIdentityValidators', I find the 'PolicyId' always 'null', I wonder where can I configure it. I guess there is some relation with 'xxxx.Config.xml', but how to set it?

AlinMoldovean, 2018-05-23:
Hi @diverger ,

You should be able to set it in the config.xml file like this

`    
    <UserTokenPolicies>
      
      <ua:UserTokenPolicy>
        <ua:TokenType>Anonymous_0</ua:TokenType>
        <ua:SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#None</ua:SecurityPolicyUri>
      </ua:UserTokenPolicy>

      
      <ua:UserTokenPolicy>
        <ua:TokenType>UserName_1</ua:TokenType>
        
        <ua:SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256</ua:SecurityPolicyUri>
      </ua:UserTokenPolicy>

      
      <ua:UserTokenPolicy>
        <ua:PolicyId>PolicyIdValue</ua:PolicyId>
        <ua:TokenType>Certificate_2</ua:TokenType>
        
        <ua:SecurityPolicyUri>http://opcfoundation.org/UA/SecurityPolicy#Basic256</ua:SecurityPolicyUri>
      </ua:UserTokenPolicy>
      
    </UserTokenPolicies>`

diverger, 2018-05-31:
I've tried that, no use. The 'TokenType' is right, but the 'PolicyId' field is null.

diverger, 2018-05-31:
Thanks, it works.

closed 2018-05-31 by diverger