OPCFoundation/UA-.NETStandard #452: Token creation takes long time [Question] ?

zlatinejc, 2018-05-22:
I am using a ```Quickstarts.DataAccessClient``` sample application to access the data periodicly from a ```OPCUA DA``` / SCADA server.

The problem here is that a simple connection to a server and data read takes long time.

After close debug I found that when my code enters the

```application.LoadApplicationConfiguration(false);```

it always takes 1 min 7 seconds + additional time to read the data.

The System configuration is:
- Windows Server 2008 R2 SP1
- .NET 4.5.1 (378758)

After examining system resources I found that the RSA process is active during this period:

[image]

Does anyone have any idea ?

The log I get:

```
PID:16292 ************************* Logging started at 05/22/2018 09:41:35 *************************
16292 - 09:42:40.801 Token #0 created. CreatedAt = 07:42:40.800 . Lifetime = 3600000
16292 - 09:42:40.895 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.336.0] [ID=390732109] Connected To: opc.tcp://SCADA:63840/ [None/None/Binary]
16292 - 09:42:40.898 Token #1 activated. CreatedAt = 07:42:40.800 . Lifetime = 3600000
16292 - 09:42:40.967 TCPCLIENTCHANNEL SOCKET CLOSED: 000003C4, ChannelId=390732109
16292 - 09:42:41.135 Token #0 created. CreatedAt = 07:42:41.135 . Lifetime = 3600000
16292 - 09:42:41.136 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.336.0] [ID=390732110] Connected To: opc.tcp://SCADA:63840/ [None/None/Binary]
16292 - 09:42:41.137 Token #1 activated. CreatedAt = 07:42:41.135 . Lifetime = 3600000
16292 - 09:42:41.155 Revised session timeout value: 60000. 
16292 - 09:42:41.155 Max response message size value: 4194304. Max request message size: 16777216 
16292 - 09:42:41.159 Server signature is null or empty.
16292 - 09:42:41.183 Empty results were received for the ActivateSession call.

```

zlatinejc, 2018-05-22:
I have an identical SCADA system set up on a different environment:

- Windows Server 2012 R2 Standard
- .NET 4.7.1 (461310)

Suprisingly the process takes 3 seconds. What ? 

Log I get:

```
PID:412 ************************* Logging started at 05/22/2018 09:18:59 *************************
412 - 09:19:02.000 Token #0 created. CreatedAt = 07:19:01.999 . Lifetime = 3600000
412 - 09:19:02.046 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.336.0] [ID=891237806] Connected To: opc.tcp://SCADA:62840/ [None/None/Binary]
412 - 09:19:02.049 Token #1 activated. CreatedAt = 07:19:01.999 . Lifetime = 3600000
412 - 09:19:02.087 TCPCLIENTCHANNEL SOCKET CLOSED: 000004EC, ChannelId=891237806
412 - 09:19:02.175 Token #0 created. CreatedAt = 07:19:02.175 . Lifetime = 3600000
412 - 09:19:02.179 SECURE CHANNEL CREATED [TcpClientChannel UA-TCP 1.02.336.0] [ID=891237807] Connected To: opc.tcp://SCADA:62840/ [None/None/Binary]
412 - 09:19:02.181 Token #1 activated. CreatedAt = 07:19:02.175 . Lifetime = 3600000
412 - 09:19:02.190 Revised session timeout value: 60000. 
412 - 09:19:02.192 Max response message size value: 4194304. Max request message size: 16777216 
412 - 09:19:02.195 Server signature is null or empty.
412 - 09:19:02.208 Empty results were received for the ActivateSession call.
```

AlinMoldovean, 2018-05-23:
Hi @zlatinejc ,

You need to debug inside the LoadApplicationConfiguration() method and find the source of the delay.

mregen, 2018-06-06:
Hi @zlatinejc are you using the quickstart client of the  UA.Net legacy (https://github.com/OPCFoundation/UA-.NET-Legacy) stack? The stack in this repo doesn't support .Net 4.5.1 and the log indicates `TcpClientChannel UA-TCP 1.02.336.0`  --> please open the issue in the legacy repo.

Anyway, LoadApplicationConfiguration searches for a matching app cert which may be the root cause for the delay if there are many certs in the store.

closed 2018-07-26 by AlinMoldovean