OPCFoundation/UA-.NETStandard #454: Documentation/simple example

btrepp, 2018-05-25:
It seems super difficult to figure out what is required to create a simple connection to a OpcUa server (eg using this library to construct a client)

Even attempting to create a simple "Client.Session" seems to require a heap of knowledge. It depends on
- ApplicationConfiguration
- Configured Endpoint
- updateBeforeConnect
- sessionName
- sessiontimeout
- identity
- locales

These aren't really described/documented, and it goes down through tree-likeness. As trying to build an ApplicationConfiguration only has constructors that appear to depend on an app.config file? Why?. Inspecting that seems to have a lot of fields that may/may not be necessary for the client side?.

This also happens with ConfiguredEndpoint, as it seems to mainly want constructors from other types. 

The consoleclient sample seems the closest, but it does through a heap of nulls and assume lots of configuration loading defaults, it may be worth writing comments in this file to it can document what assumptions this library has/requires to be used.

viados, 2018-05-25:
@btrepp This is why - especially you're not interested in such detailed knowledge - many people work with commercial solutions from one of the OPC Foundation members. I've made some very good experiences with the **advanced OPC UA Framework from Traeger**. Others work with frameworks from Unified Automation or Softings. Maybe you're interested in the solution from Traeger, they have a bunch of short how to snippets on their Wiki as well: **[Traeger OPC UA Client Development Guide](https://wiki.traeger.de/en/software/sdk/opc-ua/opc-ua.framework.advanced/development.guide.client)**

A OPC UA Client can be created using a few lines of code (as described on their page):

`using (var client = new OpcClient("opc.tcp://localhost:4840/")) {
    client.Connect();
    // Your code to interact with the server.
}`

btrepp, 2018-05-25:
Ah okay. I was planning on shimming open62541 to use as a OPCUA client for .NET, but I found this one and thought it would be nicer to use the official SDK.

Is this not considered a production ready SDK yet?.

viados, 2018-05-25:
I don't know any reason why you shouldn't use the offical SDK directly from GitHub. Just **ensure that your project meets the license terms** associated with the source shared here.

> Is this not considered a production ready SDK yet?.
I'm not sure, but commerical members additionally ensure some basic functionality on their products (internally using the stack of the foundation). Therefore using a commerical product may be a better code base to start.

mijosch, 2019-03-29:
This is still a problem!

ospani, 2020-10-08:
After all this time, the lack of any sort of documentation from the seemingly _official_ creators is one of the reasons we cannot even start to implement OPC UA in our business.

puppetSpace, 2022-05-13:
> After all this time, the lack of any sort of documentation from the seemingly _official_ creators is one of the reasons we cannot even start to implement OPC UA in our business.

same here. There is no documentation on what the classes do or why they are needed. And the documentation that's available can't really be classified as documentation.