OPCFoundation/UA-.NETStandard #458: I don't know the original reason of 'BadNothingTodo'

ggokka, 2018-05-31:
Write: Write a VQT, include VQ.
Write.Response.ResponseHeader.ServiceResult is Bad: BadNothingToDo (0x800f0000)

[image]

AlinMoldovean, 2018-05-31:
@ggokka ,

You need to check in the server where the error code is returned.
Normally that is because the service request does not contain any operation.

ggokka, 2018-05-31:
@AlinMoldovean One of the source is below

        public override ResponseHeader RegisterNodes(
            RequestHeader requestHeader, 
            NodeIdCollection nodesToRegister, 
            out NodeIdCollection registeredNodeIds)
        {
            registeredNodeIds = null;

            OperationContext context = ValidateRequest(requestHeader, RequestType.RegisterNodes);

            try
            {
                if (nodesToRegister == null || nodesToRegister.Count == 0)
                {
                    throw new ServiceResultException(StatusCodes.BadNothingToDo);
                }

AlinMoldovean, 2018-05-31:
In that case, the client sent a request with an empty list of nodesToRegister.
And is normal for the server to reject it.

closed 2018-05-31 by AlinMoldovean