OPCFoundation/UA-.NETStandard #459: OnCallMethod not triggerd when set in LoadPredefinedNodes

tomsoftware, 2018-06-01:
I use the code from #364 and on the result `NodeStateCollection predefinedNodes` of `LoadFromNodeSet2` in my I `CustomNodeManager2` I set the `OnCallMethod` for my Method-Node but it never gets triggerd. What I'm doing wrong?

```
private XmlElement[] LoadFromNodeSet2Xml(ISystemContext context, string filePath, bool updateTables, List<string> namespaceUris, NodeStateCollection predefinedNodes)
{
    ...
    var result = LoadFromNodeSet2(context, istrm, updateTables, namespaceUris, predefinedNodes);

    MethodState node = predefinedNodes[1];

    node.OnCallMethod = new GenericMethodCalledEventHandler(OnStartCall);

    return result;
}
```

On the other hand if I use the `override void Call(OperationContext context,` on my `CustomNodeManager2` I resive the call from the client.

I also use the above way to set the value of `PropertyState` nodes and that is working!? Do I generally have to access the dynamic nodes of NodesManager in another way or does the behavior for `PropertyState` and `MethodState` nodes differ?

AlinMoldovean, 2018-06-04:
@tomsoftware ,

A better way would be to implement an override for _AddBehaviourToPredefinedNode()_
Something similar with implementation of DiagnosticsNodeManager for OnConditionRefresh:
```
protected override NodeState AddBehaviourToPredefinedNode(ISystemContext context, NodeState predefinedNode)
        {
            BaseObjectState passiveNode = predefinedNode as BaseObjectState;

            if (passiveNode == null)
            {
                MethodState passiveMethod = predefinedNode as MethodState;

                if (passiveMethod == null)
                {
                    return predefinedNode;
                }

                if (passiveMethod.NodeId == MethodIds.ConditionType_ConditionRefresh)
                {
                    ConditionRefreshMethodState activeNode = new ConditionRefreshMethodState(passiveMethod.Parent);
                    activeNode.Create(context, passiveMethod);

                    // replace the node in the parent.
                    if (passiveMethod.Parent != null)
                    {
                        passiveMethod.Parent.ReplaceChild(context, activeNode);
                    }

                    activeNode.OnCall = OnConditionRefresh;

                    return activeNode;
                }

                return predefinedNode;
            }

            NodeId typeId = passiveNode.TypeDefinitionId;

            if (!IsNodeIdInNamespace(typeId) || typeId.IdType != IdType.Numeric)
            {
                return predefinedNode;
            }

            switch ((uint)typeId.Identifier)
            {
                case ObjectTypes.ServerType:
                    {
                        if (passiveNode is ServerObjectState)
                        {
                            break;
                        }

                        ServerObjectState activeNode = new ServerObjectState(passiveNode.Parent);
                        activeNode.Create(context, passiveNode);

                        // add the server object as the root notifier.
                        AddRootNotifier(activeNode);

                        // replace the node in the parent.
                        if (passiveNode.Parent != null)
                        {
                            passiveNode.Parent.ReplaceChild(context, activeNode);
                        }

                        return activeNode;
                    }

            }

            return predefinedNode;
        }
```

tomsoftware, 2018-06-05:
Hi @AlinMoldovean 

thanks again but the client still returnes
 `Call failed with status 'BadArgumentsMissing'`
when calling my UAMethod and the OnStartCall is not triggered.

```
protected override NodeState AddBehaviourToPredefinedNode(ISystemContext context, NodeState predefinedNode)
{
  BaseObjectState passiveNode = predefinedNode as BaseObjectState;

  if (passiveNode == null)
  {
    MethodState passiveMethod = predefinedNode as MethodState;

    if (passiveMethod != null)
    {
      passiveMethod.OnCallMethod = new GenericMethodCalledEventHandler(OnStartCall);
      return passiveMethod;
    }
  }

  return predefinedNode;
}
```

AlinMoldovean, 2018-06-06:
@tomsoftware ,
Please check if the ConditionRefresh method works and and to make the same implementation on your NodeManager class

tomsoftware, 2018-06-08:
OK, I now create a new Instance of MethodState and overwrite the given NodeState of the not working MethodState in AddBehaviourToPredefinedNode. Now I'm able to change/set the OnCallMethod. 

I think the generated NodeState of the instance, generated out of the xml-defined-type has some issues not allowing me to setting the OnCallMethod function.

Thanks for your help

closed 2018-06-08 by AlinMoldovean