OPCFoundation/UA-.NETStandard #460: Finding Node on Server side

tomsoftware, 2018-06-01:
How to use `TranslateBrowsePath` or `Find` on Serverside to find a NodeId by its browsename.

```
class MyServer : StandardServer {
  ...
  var result= this.ServerInternal.CoreNodeManager.Find(Opc.Ua.ObjectIds.ObjectsFolder, "Server");
}
```
result will by null but 
`var result= this.ServerInternal.CoreNodeManager.Find(Opc.Ua.ObjectIds.ObjectsFolder, "");`
will return the ObjectFolder

or do I need to use the  `NodeManager.TranslateBrowsePathsToNodeIds` but what to use as OperatinContext on the server -> OperationContext (`Server.DefaultSystemContext.OperationContex` ) fails with an NullReferenceException exception cause DefaultSystemContext.OperationContext == null. 

thank you

AlinMoldovean, 2018-06-04:
@tomsoftware ,

FindPredefinedNode method of NodeManager can be used for searching nodes, but it searches by NodeId. 
e.g. 
```
NodeState state = Server.DiagnosticsNodeManager.FindPredefinedNode(NodeId, null);
```

An option ca be to search inside PredefinedNodes, but consider that you can have multiple results (i.e. nodes with the same browse name) in a NodeManager.

tomsoftware, 2018-06-05:
@AlinMoldovean: Thanks so far,

So `NodeManager.TranslateBrowsePathsToNodeIds` is not allowed on Serverside or do I need to generate my own `OperatinContext` on the Server cause `DefaultSystemContext.OperationContext == null`?

The problem is that I'm struggling with many dynamic nodes, created on runtime and sometimes need to be accessed by it's browse name by the server.

AlinMoldovean, 2018-06-06:
@tomsoftware ,

TranslateBrowsePath implementation is intended to handle service requests from a client session.
You can try to make a method with a similar implementation without need of OperationContext or you can search inside PredefinedNodes of the NodeManager

tomsoftware, 2018-06-08:
Thanks again

now I use a recursively function to "jump" from node to node by reading the nodes references with GetReferences() and comparing the BrowseName myself. 

This looks good but sometimes I don't get any reference from a node (NodeState)? Why is that? Is there a global "reference"-Manager cause from the OPC-UA Client I can see the references or do I need to query the references in a special way?

AlinMoldovean, 2018-06-08:
@tomsoftware ,

In that case the nodes can be the target of a one-way reference.
I recommend again to loop inside the PredefinedNodes list of your NodeManager and search for required BrowseName instead of making a recursive browse.

tomsoftware, 2018-06-11:
@AlinMoldovean Thanks for help.
It seems I was on the wrong path. I now use `NodeState.CreateBrowser ` recursively to jump from node to node with the `ServerInternal.DefaultSystemContext`.

My problem was that I queried many different NameSpaces and on some NodeManager I was unable to query the PredefinedNodes List. So CreateBrowser does this the right way. For everyone who is also on how to resolve a browsepath on serverside:

```
private CustomNodeManager2 GetNodeManagerFromNodeId(NodeId nodeId)
{
	INodeManager nodeManager = null;
	masterNodeManager.GetManagerHandle(nodeId, out nodeManager);

	return nodeManager as CustomNodeManager2;
}

private NodeState GlobalFindNode(NodeId nodeId)
{
	CustomNodeManager2 nm = GetNodeManagerFromNodeId(nodeId);
	if (nm == null) return null;
	return nm.Find(nodeId);
}

public NodeId PathToNodeId(NodeId startNode, string[] path, int pathIndex)
{
	if (pathIndex >= path.Length) return startNode;

	/// split your path[pathIndex] to namespaceIndexToFind and browseNameToFind ...

	NodeState source = GlobalFindNode(startNode);
	if (source == null) return null;

	ContinuationPoint continuationPoint = new ContinuationPoint();

	/// see: TranslateBrowsePath @ \Opc.Ua.Server\Diagnostics\CustomNodeManager.cs
	// create a new browser.
	INodeBrowser browser = source.CreateBrowser(
               yourServer.ServerInternal.DefaultSystemContext,
               null,
               continuationPoint.ReferenceTypeId,
               continuationPoint.IncludeSubtypes,
               continuationPoint.BrowseDirection,
               null,
               null,
               false);

	lock (browser)
	{
		for (IReference reference = browser.Next(); reference != null; reference = browser.Next())
		{
			if (reference.IsInverse) continue;

			if ((reference.ReferenceTypeId == ReferenceTypeIds.HasComponent)
				|| (reference.ReferenceTypeId == ReferenceTypeIds.HasProperty)
				|| (reference.ReferenceTypeId == ReferenceTypeIds.Organizes)
				|| (reference.ReferenceTypeId == ReferenceTypeIds.HasSubtype))
			{
				if ((reference.TargetId.NamespaceIndex != namespaceIndexToFind) && (namespaceIndexToFind >= 0)) continue;

				NodeId targetNodeId = new NodeId(reference.TargetId.Identifier, reference.TargetId.NamespaceIndex);

				CustomNodeManager2 targetNodeManager = GetNodeManagerFromNodeId(targetNodeId);
				if (targetNodeManager == null) continue;

				var s = targetNodeManager.Find(targetNodeId);

				if (string.Compare(s.BrowseName.Name, browseNameToFind, false) == 0)
				{
				    return PathToNodeId(targetNodeId, path, pathIndex + 1);
				}
			}
		}
	}

	/// not found
	return null;
}
```

closed 2018-06-11 by tomsoftware

rudoq32, 2020-03-29:
Hello,
is there a straight forward way to Find a node by string id using OPCFoundation nuget in MSVS C#?
I am getting null for node in code 
```
NodeId nodeId = new NodeId("ns=0;s=Server");
INode node =  session.NodeCache.Find(nodeId);
```