OPCFoundation/UA-.NETStandard #466: Certificate validity period should be set from call to "CheckApplicationInstanceCertificate"

CrustyApplesniffer, 2018-06-11:
The ```ApplicationInstance.CheckApplicationInstanceCertificate``` function should be overridden to allow the UA client creating a certificate with a specific ```LifeTimeInMonth``` value.

For example, to create a certificate with a validity period of 24 months,

```
var application = new ApplicationInstance
            {
                ApplicationName = applicationName,
                ApplicationType = ApplicationType.Client,
                ApplicationConfiguration = config
            };           
            application.CheckApplicationInstanceCertificate(false, 1024, 24).GetAwaiter().GetResult();
```
instead of 
```
var application = new ApplicationInstance
            {
                ApplicationName = applicationName,
                ApplicationType = ApplicationType.Client,
                ApplicationConfiguration = config
            };
            application.CheckApplicationInstanceCertificate(false, 1024).GetAwaiter().GetResult();
```

kvloover, 2018-06-13:
It would indeed be nice to have this option so this can be matched with planned maintenance jobs and/or life of products, etc...

For now you can set the certificate manually. You can create this certificate using the CertificateFactory
```
config.SecurityConfiguration.ApplicationCertificate.Certificate = CertificateFactory.CreateCertificate(
                                                config.SecurityConfiguration.ApplicationCertificate.StoreType,
                                                config.SecurityConfiguration.ApplicationCertificate.StorePath,
                                                null,
                                                config.ApplicationUri,
                                                config.ApplicationName,
                                                config.SecurityConfiguration.ApplicationCertificate.SubjectName,
                                                null,
                                                CertificateFactory.defaultKeySize,
                                                DateTime.UtcNow - TimeSpan.FromDays(1),
                                                24,//CertificateFactory.defaultLifeTime,
                                                CertificateFactory.defaultHashSize,
                                                false,
                                                null,
                                                null
                                                );
```

CrustyApplesniffer, 2018-06-13:
Thank you, 
I'll try this.

CrustyApplesniffer, 2018-06-14:
It works.

closed 2018-06-14 by CrustyApplesniffer