OPCFoundation/UA-.NETStandard #473: Split the nuget package into a client part and a server part

CrustyApplesniffer, 2018-06-14:
I think it will be wise to split the ```OPCFoundation.NetStandard.Opc.Ua``` nuget package into 2 parts:
- The client one
- The server one

In fact when you develop a system that will use OPC UA, each project needs only one of theses parts.

The client part will be compound by
- Opc.Ua.Client
- Opc.Ua.Core
- Opc.Ua.Configuration and will reference all packages this 3 libraries depend on.

The server part will be compound by 
- Opc.Ua.Server
- Opc.Ua.Core
- Opc.Ua.Configuration  and their dependencies.

It will reduce the dependencies list.

AlinMoldovean, 2018-06-14:
HI @CrustyApplesniffer ,

We are aware of this overhead, but for the moment we want to keep a single NuGet package for all components in order to simplify things.

flts, 2018-07-11:
Wouldn't it be possible to have this 'All' NuGet package which contains all the components like it is now and next to this 'All' NuGet Package there will be 'Client' and 'Server' NuGet Packages.

- OPCFoundation.NetStandard.Opc.Ua _(in the future this could maybe be transitioned to OPCFoundation.NetStandard.Opc.Ua.All)_
- OPCFoundation.NetStandard.Opc.Ua.Client
- OPCFoundation.NetStandard.Opc.Ua.Server

Because the OPCFoundation.NetStandard.Opc.Ua NuGet package currently contains all the components we are unable to use it. I have had to download the sources, recompile it locally and distribute the correct assemblies manually in the projects we are using library.

I could look into this and create a PR if you would like to.

AlinMoldovean, 2018-10-25:
The decision is to provide a single Nuget package for this repository.

closed 2018-10-25 by AlinMoldovean