OPCFoundation/UA-.NETStandard #475: How to find the variable(s) involved in a FastDataChangeCallback ?

CrustyApplesniffer, 2018-06-15:
Hello,

I create a subscription with several items

```
var subscription = new Subscription(_session.DefaultSubscription) { PublishingInterval = 1000 };

foreach (var opcItem in items)
{
    var monitoredItem = new MonitoredItem(subscription.DefaultItem) { StartNodeId = new NodeId($"ns=2;s={opcItem}"),DisplayName = opcItem};    
    subscription.AddItem(monitoredItem);
}

_session.AddSubscription(subscription);
subscription.FastDataChangeCallback += FastDataChangeCallback;
subscription.Create();
```

with the callback defined as follow:
```
private void FastDataChangeCallback(Subscription subscription, DataChangeNotification notification, IList<string> stringtable)
```

When the ```FastDataChangeCallback``` raises, how can I find which element of ```subscription.MonitoredItems``` matches a ```notification.MonitoredItems``` element ?

I want the DisplayName of each element of ```notification.MonitoredItems```

CrustyApplesniffer, 2018-06-15:
I'm using the ClientHandle but I'm not sure its the right/best way:

```
private void FastDataChangeCallback(Subscription subscription, 
                                    DataChangeNotification notification,
                                    IList<string> stringtable)
{
   
    foreach (var value in notification.MonitoredItems)
    {
        var handle = value.ClientHandle;
        var node = subscription.MonitoredItems.FirstOrDefault(n => n.ClientHandle == handle);

        var typeInfo = TypeInfo.Construct(value.Value.Value);
        var current = TypeInfo.Cast(value.Value.Value, typeInfo, typeInfo.BuiltInType);     

        System.Diagnostics.Debug.WriteLine("[UA] " + node?.DisplayName + " " +
                                           value.Value.ServerTimestamp + ": " + current);
    }
}
```

AlinMoldovean, 2018-06-19:
Hello @CrustyApplesniffer ,

You can identify the local MonitoredItem from the "notification" object using by "notification.ClientHandle".
You code seems to be correct.

closed 2018-07-26 by AlinMoldovean