OPCFoundation/UA-.NETStandard #480: Using a configuration file with a name different of the section name raises an exception

CrustyApplesniffer, 2018-06-19:
In my app.config file I've set :
```
  <configSections>
    <section name="OpcUaServer" type="Opc.Ua.ApplicationConfigurationSection,Opc.Ua.Core" />
  </configSections>

  <OpcUaServer>
    <ConfigurationLocation xmlns="http://opcfoundation.org/UA/SDK/Configuration.xsd">
      <FilePath>OPC_UA_Server.Config.xml</FilePath>
    </ConfigurationLocation>
  </OpcUaServer>  
```

the ```OPC_UA_Server.Config.xml``` should be used as configuration file.
In fact, the ```GetFilePathFromAppConfig``` method returns OpcUaServer.config.xml as configuration file which doesn't exist and raises a ```BadConfigurationError: Could not load configuration file```

```
public static string GetFilePathFromAppConfig(string sectionName)
{
    // convert to absolute file path (expands environment strings).
    string absolutePath = Utils.GetAbsoluteFilePath(sectionName + ".Config.xml", true, false, false);
    if (absolutePath != null)
    {
        return absolutePath;
    }
    else
    {
        return sectionName + ".Config.xml";
    }
}
```

CrustyApplesniffer, 2018-06-19:
In fact I've set application.ConfigSectionName to ```OpcUaServer```.
The values set in the app.config file seem not to be used.

AlinMoldovean, 2018-07-26:
It seems that `GetFilePathFromAppConfig(string sectionName) `returns only` sectionName + ".Config.xml"` without reading the app.config file.
The problem was cased by missing support for `System.Configuration.ConfigurationManager` when porting the code to NET Standard.

closed 2018-10-25 by AlinMoldovean