OPCFoundation/UA-.NETStandard #482: Reading/Writing tags using the UA Universal Windows sample client

mindovrmech, 2018-06-20:
I'm relatively new to the Universal Windows Platform, and I'm trying to use the UA Universal Windows Opc.Ua.SampleClient project as an example to figure out how to set up an OPC client and read/write tags.  I've built the project, and I can connect to an OPC server (on a Siemens PLC).  However, I can't seem to use the sample client program to read or write to tags.  I can browse the tag tree on the OPC server, and I found the "Report" button that allows me to place a selected tag onto a subscription list, but no read or write functionality.
I was able to build the .NET Standard version of the OPC Sample Client, and in that version you right click on a tag in the tag browser and it gives you the option to read or write to the selected tag.
Am I just missing some basic user interface thing in Universal Windows Platform?

AlinMoldovean, 2018-10-25:
UWP sample is just a prof of concept and it has  a limited set of features.

closed 2018-10-25 by AlinMoldovean

ga5tan, 2019-06-23:
@mindovrmech how did u connect to Siemens PLC? It wont show in Discovery :(

mindovrmech, 2019-06-24:
In that particular case, I was using the SimaticNET OPC server (a software
package from Siemens) to bring the S7-1200 PLC data into the OPC world.
Since then, I've also had success using Ignition (a software package from
Inductive Automation) to connect an S7-1200 PLC to an OPC server.

ga5tan, 2019-06-25:
@mindovrmech thx, I guess my primary problem is, I am trying UA client from OPCFundation to talk to old OPC Classic servers :(