OPCFoundation/UA-.NETStandard #483: How to correctly activate session

pFonta88, 2018-06-21:
Good morning,

How can I obtain the SignatureData object to be passed as parameter in ActivateSession() call?

I'm trying with this code:

` SignatureData clientSignature = SecurityPolicies.Sign(opcUaClientConfiguration.SecurityConfiguration.ApplicationCertificate.Certificate, opcUaSession.ConfiguredEndpoint.Description.SecurityPolicyUri, dataToSign);`

but I don't understand where can I found the byte[] dataToSend.

Can I have some precisations about this and in general about ActivateSession() method?

Thanks,
Paolo

AlinMoldovean, 2018-07-03:
@pFonta88 ,

You do not need to put application code for ActivateSession() call for creating a session.
You can use Session.Create() method.
It will invoke all the required OPC UA service calls for creating a session, i.e. OpenSecureChannel, CreateSession and ActivateSession.

pFonta88, 2018-07-03:
@AlinMoldovean 

I'm trying to do this because reconnect() don't works for me, so I thinked that maybe my session wasn't completely well activated in order to allow reconnection.

My code, at the moment create the session with Session.Create() as you suggested, but when the connection is lost, reconnect() cannot restore it. The software remains in wait of a successuful reconnect but when it happens it don't receive values anymore.

Have you any suggestion about how to configure correctly an auto reconnection?
Is there anything to check machine-side to ensure that it can work?

Thanks,
Paolo

AlinMoldovean, 2018-07-03:
@pFonta88 ,

For session reconnect on client side please check the implementation of `SessionReconnectHandler `class

eltigre4jc, 2018-07-04:
I subscribe to what @pFonta88  says, Reconnect does not work correctly or at all.

If it reconnects, when I try to browse, I get this error:

```
Cannot access a disposed object.
Object name: 'TransportChannel is not available.'.
---    at Opc.Ua.ClientBase.get_UseTransportChannel()
   at Opc.Ua.SessionClient.Browse(RequestHeader requestHeader, ViewDescription view, UInt32 requestedMaxReferencesPerNode, BrowseDescriptionCollection nodesToBrowse, BrowseResultCollection& results, DiagnosticInfoCollection& diagnosticInfos)
   at Opc.Ua.Client.Session.Browse(RequestHeader requestHeader, ViewDescription view, NodeId nodeToBrowse, UInt32 maxResultsToReturn, BrowseDirection browseDirection, NodeId referenceTypeId, Boolean includeSubtypes, UInt32 nodeClassMask, Byte[]& continuationPoint, ReferenceDescriptionCollection& references)
   at WindowsApp1.OPC.Browse(Session& session, ExpandedNodeId nodeId, Boolean isFirstTime)
```

Also tried to remake the whole session, but without success

AlinMoldovean, 2018-10-25:
I tested the reconnect mechanism from UA Sample Client and it worked fine.
Please use the example from  SampleApplications\Samples\Controls.Net4\ClientForm.cs

closed 2018-10-25 by AlinMoldovean