OPCFoundation/UA-.NETStandard #484: Properly disposing sessions

AStove, 2018-06-21:
When the connection to a device is lost I close and dispose of a Session and then start a new one until it connects. However multiple session instances are alive and have an open TCP connection after this. For some reason the Sessiosn are not properly disposed of.

`_session.KeepAlive -= KeepAliveEventHandler;`
`_session.Close(500);`
`_session.Dispose();`
`_session = null;`

[image]

wemgiraldo, 2018-06-27:
How did you solved this problem?

AStove, 2018-06-29:
I stopped doing it in a seperate thread and it seems to behave better now. I'm sorry I don't have a clear example of where it went wrong. But to me if you dispose something and then cancel the thread, it should be GCed and it didn't. It even kept TCP connections open.

AStove, 2018-06-29:
I just noticed I have two referenced to it still. More than probably my fault anyway.

closed 2018-07-26 by AlinMoldovean