OPCFoundation/UA-.NETStandard #486: OPC UA Server - String Node Id Creation

DummyRick, 2018-06-23:
Hi evreyone,

I'm trying to create an OPC UA Server that contains some methods and I need to create a NodeId for each method i want expose.
Using Integer identifier it's works but with String identifier i'm not able to call the method using an OPC UA client.

Sample client return this error:

[image]

Trying with a 3th part client still not work but the error is missing NodeId, it's seem like a numeric node id it's expected.

I'm not sure if it's a bug or just something that escape me. Can someone please clarify my doubts? Thanks,

NodeId integer created using:

> 
>         #region public NodeId(uint value, ushort namespaceIndex)
>         /// <summary>
>         /// Initializes a guid node identifier with a namespace index.
>         /// </summary>
>         /// <remarks>
>         /// Creates a new NodeId that will use a numeric (unsigned int) for its Id, but also
>         /// specifies which namespace this node should belong to.
>         /// </remarks>
>         /// <param name="value">The new (numeric) Id for the node being created</param>
>         /// <param name="namespaceIndex">The index of the namespace that this node should belong to</param>
>         /// <seealso cref="SetNamespaceIndex"/>
>         public NodeId(uint value, ushort namespaceIndex)
>         {
>             m_namespaceIndex = namespaceIndex;
>             m_identifierType = IdType.Numeric;
>             m_identifier = value;
>         }
> 
>         #endregion

NodeId string created using:
   

>          #region public NodeId(string value, ushort namespaceIndex)
>         /// <summary>
>         /// Initializes a string node identifier with a namespace index.
>         /// </summary>
>         /// <remarks>
>         /// Creates a new NodeId that will use a string for its Id, but also
>         /// specifies if the Id is a URI, and which namespace this node belongs to.
>         /// </remarks>
>         /// <param name="value">The new (string) Id for the node being created</param>
>         /// <param name="namespaceIndex">The index of the namespace that this node belongs to</param>
>         public NodeId(string value, ushort namespaceIndex)
>         {
>             m_namespaceIndex = namespaceIndex;
>             m_identifierType = IdType.String;
>             m_identifier     = value;
>         }
> 
>         #endregion

fchaxel, 2018-06-24:
Hi,

I have something like

` MethodState method = new MethodState(parent);
method.NodeId="Add device",1);`

also method arguments are identified with a String.

It's working perfectly with OpcUaExpert & a C# client. So the error is certainly not due to that.

Bye.

AlinMoldovean, 2018-06-27:
Hi @DummyRick ,

Please check in the implementation of ReferenceServer project.
 ReferenceNodeManager.CreateMethod(NodeState parent, string path, string name)
creates method nodes with string Node identifiers.

```
private MethodState CreateMethod(NodeState parent, string path, string name)
        {
            MethodState method = new MethodState(parent);

            method.SymbolicName = name;
            method.ReferenceTypeId = ReferenceTypeIds.HasComponent;
            method.NodeId = new NodeId(path, NamespaceIndex);
            method.BrowseName = new QualifiedName(path, NamespaceIndex);
            method.DisplayName = new LocalizedText("en", name);
            method.WriteMask = AttributeWriteMask.None;
            method.UserWriteMask = AttributeWriteMask.None;
            method.Executable = true;
            method.UserExecutable = true;

            if (parent != null)
            {
                parent.AddChild(method);
            }

            return method;
        }
```

closed 2018-07-26 by AlinMoldovean