OPCFoundation/UA-.NETStandard #490: Triggering a ModelchangeEvent

SirWolf66, 2018-06-27:
Hi,

For an internship assignment i need to update the UI of a client containing the Nodes of the server Information Model, if a Node gets added or removed from the information model on the server. For this i figured the client needs to subscribe to the GeneralModelchangeEvent on the server Object.

It is mentioned in Issue  #208 that "For the moment this feature is not implemented in the .NET Standard Server library, so the server is not automatically triggering ModelChange events."
The question i have is, if this is still the case and if so, i presume i need to trigger the GeneralModelChange manually. Which brings me to my second question: how to trigger a GeneralModelchange event manually?

Also, just to be sure, Client side a subscription is made, which contains a monitored item for the Eventnotifier Attribute of the Server Object. This would subscribe to all Events which are triggered, including the GeneralModelChangeEvent, right?

AlinMoldovean, 2018-07-03:
@SirWolf66 ,

You need to use the Server.ReportEvent() method for triggering events from server.
En example can be found in AlarmConditionNodeManager.OnRaiseSystemEvents()

fchaxel, 2018-07-04:
Hi @SirWolf66 

`

                // Un event pour signaler la modification du fodler
                GeneralModelChangeEventState ev = new GeneralModelChangeEventState(RabbitRoot);
                ev.Initialize(SystemContext,RabbitRoot,EventSeverity.Medium,new LocalizedText("Model change"));

                ev.Changes = new PropertyState<ModelChangeStructureDataType[]>(null);
                ev.Changes.Value = new ModelChangeStructureDataType[] { new ModelChangeStructureDataType()};

                ev.Changes.Value[0].Affected = RabbitRoot.NodeId;
                ev.Changes.Value[0].AffectedType = ObjectTypeIds.FolderType;
                ev.Changes.Value[0].Verb = 4; // 4 RefAdd, 8 RefDelete : 

                ev.SetChildValue(SystemContext, BrowseNames.SourceNode, RabbitRoot.NodeId, false);
                ev.SetChildValue(SystemContext, BrowseNames.SourceName, RabbitRoot.DisplayName, false);
                Server.ReportEvent(ev);
`

where RabbitRoot here is a FolderState where a modification (something added) as been done.

Bye

fchaxel, 2018-07-04:
On my side it is called during a remote method call (an Opc client call a method to add new items into the address space, then after AddPredefinedNode and before 'return ServiceResult.Good;' the ReportEvent is used).

SirWolf66, 2018-07-04:
@fchaxel, thank you that indeed works!

closed 2018-10-25 by AlinMoldovean